dtcooper/raspotify #172: Please help - Raspberry PI zero + bluetooth speaker, no sound and can't find setup that works

aintnogeek, 2018-12-28:
Hi,

I have found two similar issues but can't find setup that works for me.

I have raspberry pi zero (with rasbian) and bt speaker , that works with for example internet radio but raspotify doesn't work.

Raspotify works with hdmi but not with bluetooth.

a)Could you make some changes in default config file to select active output depending on what is selected in rasbian as audio output?
b) Please could someone create dummy guide step by step how to make raspotify sound output to bluetooth speaker?

Thank you!

fosterbuster, 2019-01-04:
Have you checked

https://github.com/dtcooper/raspotify/issues/63 ?

aintnogeek, 2019-01-12:
Yes it worked at the end.

Important thing is not to connect BT speaker trough UI but with bluethootctrl.

and also make adjustments in file as in post you mentioned

xtools-at, 2020-03-28:
had a similar issue, installing this alongside (and removing pulseaudio if you have it) resolved it for me: https://github.com/bablokb/pi-btaudio

vguttmann, 2020-07-17:
If the issue is resolved, you should close it @aintnogeek

closed 2021-11-17 by JasonLG1979