dtcooper/raspotify #173: Possibility of restarting raspotify

janekszynal, 2019-01-05:
Hey, thanks so much for this project, it does really Just Work™ most of the time. I did, however, have some small issues, in particular when trying to run an AirPlay server on my pi at the same time. It's hard to debug, because every time I want raspotify to restart, I reboot my pi. I'm sure there has to be a softer way, but I wasn't able to find it on my own.

kosciCZ, 2019-01-05:
I'm confident this is the command you're looking for:
`sudo systemctl restart raspotify`
As mentioned in the README, it's used to restart raspotify after changing config.

janekszynal, 2019-01-05:
Thanks a lot, works perfectly. I did (of course) go to the readme for info about this, but failed to notice this line. @dtcooper you could consider putting it in a separate section, but it could very well have been just me.

closed 2019-01-05 by janekszynal