dtcooper/raspotify #176: Switching songs causes Raspotify to break

Squidman474, 2019-01-22:
Whenever I switch songs quickly (i.e. not waiting for 5-10 seconds after switching each song), raspotify breaks. Music stops playing entirely at first, followed a few seconds later by a horrible distorted pulsing noise. From there, I can't even pull up the terminal on my pi- the entire thing freezes up. Any idea how to fix this? I don't really mind waiting a few seconds to switch a song, but when I forget occasionally, I have to unplug and restart the entire pi.

Squidman474, 2019-01-29:
I've tried uninstalling raspotify, even reformatting the sd card. No luck.

warro, 2019-02-20:
Same problem here.

Razer0123, 2019-02-22:
Same, any help would be appreciated

brookshank, 2019-02-22:
What version pi are you all running on? What kind of power supply?

knarxy, 2019-03-02:
Same problem

closed 2021-11-24 by github-actions[bot]