dtcooper/raspotify #180: Raspotify not visible in device list

vwishingrad, 2019-02-18:
Hi,

I installed the latest version of stretch on a Raspberry Pi 3. Then I installed raspotify and set it to auto start on boot `sudo systemctl enable raspotify`. I rebooted the pi and ran `sudo systemctl status raspotify` to check that raspotify started, it has. However, I cannot see it in the devices menu from laptop/phone. I have spotify premium. I've restarted the pi and client devices, but no luck. Any ideas?

xMordax, 2019-02-20:
firewall?

vwishingrad, 2019-02-20:
> firewall?

Good idea, but no firewall

xMordax, 2019-02-20:
mind sharing logs and config here?

vwishingrad, 2019-03-05:
Sure, which logs?

vwishingrad, 2019-03-06:
Here's my config file:

```
# /etc/default/raspotify -- Arguments/configuration for librespot

# Device name on Spotify Connect
DEVICE_NAME="Raspberry Pi"

# Bitrate, one of 96 (low quality), 160 (default quality), or 320 (high quality)
BITRATE="320"

# Additional command line arguments for librespot can be set below.
# See `librespot -h` for more info. Make sure whatever arguments you specify
# aren't already covered by other variables in this file. (See the daemon's
# config at `/lib/systemd/system/raspotify.service` for more technical details.)
#
# To make your device visible on Spotify Connect across the Internet add your
# username and password which can be set via "Set device password", on your
# account settings, use `--username` and `--password`.
#
# To choose a different output device (ie a USB audio dongle or HDMI audio out),
# use `--device` with something like `--device hw:0,1`. Your mileage may vary.
#
#OPTIONS="--username <USERNAME> --password <PASSWORD>"

# Uncomment to use a cache for downloaded audio files. Cache is disabled by
# default. It's best to leave this as-is if you want to use it, since
# permissions are properly set on the directory `/var/cache/raspotify'.
#CACHE_ARGS="--cache /var/cache/raspotify"

# By default, the volume normalization is enabled, add alternative volume
# arguments here if you'd like, but these should be fine.
#VOLUME_ARGS="--enable-volume-normalisation --linear-volume --initial-volume=100"

# Backend could be set to pipe here, but it's for very advanced use cases of
# librespot, so you shouldn't need to change this under normal circumstances.
#BACKEND_ARGS="--backend alsa"
```

Any help finding the logs would be appreciated!

NoXPhasma, 2019-04-14:
Log can be accessed with `journalctl -u raspotify`.

Gerald12344, 2020-01-06:
I know I am a little late and this might sound a little stupid but on the options line, the hashtag is commenting out the options so the program skips over it try removing the hashtag.

georgegeorgetj, 2020-05-04:
Hello, I got the exact same problem.
The logs are as follows:

```
gtj@nanopineoplus2:~$ journalctl -u raspotify
-- Logs begin at Mon 2020-05-04 02:26:25 UTC, end at Mon 2020-05-04 02:43:52 UTC
May 04 02:26:32 nanopineoplus2 systemd[1]: Starting Raspotify...
May 04 02:26:32 nanopineoplus2 systemd[1]: Started Raspotify.
May 04 02:26:32 nanopineoplus2 librespot[963]: [2020-05-04T02:26:32Z INFO  libre
May 04 02:26:33 nanopineoplus2 librespot[963]: thread 'main' panicked at 'called
May 04 02:26:33 nanopineoplus2 librespot[963]: stack backtrace:
May 04 02:26:33 nanopineoplus2 librespot[963]:    0:  0x10b13a4 - backtrace::bac
May 04 02:26:33 nanopineoplus2 librespot[963]:                        at /cargo/
May 04 02:26:33 nanopineoplus2 librespot[963]:    1:  0x10b13a4 - backtrace::bac
May 04 02:26:33 nanopineoplus2 librespot[963]:                        at /cargo/
May 04 02:26:33 nanopineoplus2 librespot[963]:    2:  0x10b13a4 - std::sys_commo
May 04 02:26:33 nanopineoplus2 librespot[963]:                        at src/lib
May 04 02:26:33 nanopineoplus2 librespot[963]:    3:  0x10b13a4 - <std::sys_comm
May 04 02:26:33 nanopineoplus2 librespot[963]:                        at src/lib
May 04 02:26:33 nanopineoplus2 librespot[963]:    4:  0x10d3ca4 - core::fmt::wri
May 04 02:26:33 nanopineoplus2 librespot[963]:                        at src/lib
May 04 02:26:33 nanopineoplus2 librespot[963]:    5:  0x10abff0 - std::io::Write
May 04 02:26:33 nanopineoplus2 librespot[963]:                        at src/lib
May 04 02:26:33 nanopineoplus2 librespot[963]:    6:  0x10b3954 - std::sys_commo
May 04 02:26:33 nanopineoplus2 librespot[963]:                        at src/lib
May 04 02:26:33 nanopineoplus2 librespot[963]:    7:  0x10b3954 - std::sys_commo
May 04 02:26:33 nanopineoplus2 librespot[963]:                        at src/lib
May 04 02:26:33 nanopineoplus2 librespot[963]:    8:  0x10b3954 - std::panicking
lines 1-23...skipping...
-- Logs begin at Mon 2020-05-04 02:26:25 UTC, end at Mon 2020-05-04 02:43:52 UTC. --
May 04 02:26:32 nanopineoplus2 systemd[1]: Starting Raspotify...
May 04 02:26:32 nanopineoplus2 systemd[1]: Started Raspotify.
May 04 02:26:32 nanopineoplus2 librespot[963]: [2020-05-04T02:26:32Z INFO  librespot] librespot (raspotify v0.14.0) 3672214 (2020-01-30). Built on 2020-02-16.
May 04 02:26:33 nanopineoplus2 librespot[963]: thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 19, kind: Other, message: "N
May 04 02:26:33 nanopineoplus2 librespot[963]: stack backtrace:
May 04 02:26:33 nanopineoplus2 librespot[963]:    0:  0x10b13a4 - backtrace::backtrace::libunwind::trace::hb1e9954dcc6e7c1f
May 04 02:26:33 nanopineoplus2 librespot[963]:                        at /cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.40/src/backtrace/libunw
May 04 02:26:33 nanopineoplus2 librespot[963]:    1:  0x10b13a4 - backtrace::backtrace::trace_unsynchronized::h1b198a73034a9734
May 04 02:26:33 nanopineoplus2 librespot[963]:                        at /cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.40/src/backtrace/mod.rs
May 04 02:26:33 nanopineoplus2 librespot[963]:    2:  0x10b13a4 - std::sys_common::backtrace::_print_fmt::hf691518b63d0b79d
May 04 02:26:33 nanopineoplus2 librespot[963]:                        at src/libstd/sys_common/backtrace.rs:84
May 04 02:26:33 nanopineoplus2 librespot[963]:    3:  0x10b13a4 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::he6a16924
May 04 02:26:33 nanopineoplus2 librespot[963]:                        at src/libstd/sys_common/backtrace.rs:61
May 04 02:26:33 nanopineoplus2 librespot[963]:    4:  0x10d3ca4 - core::fmt::write::hc426988e4c4119a8
May 04 02:26:33 nanopineoplus2 librespot[963]:                        at src/libcore/fmt/mod.rs:1025
May 04 02:26:33 nanopineoplus2 librespot[963]:    5:  0x10abff0 - std::io::Write::write_fmt::h34e90dde595ba7cd
May 04 02:26:33 nanopineoplus2 librespot[963]:                        at src/libstd/io/mod.rs:1426
May 04 02:26:33 nanopineoplus2 librespot[963]:    6:  0x10b3954 - std::sys_common::backtrace::_print::h406863a8800396cd
May 04 02:26:33 nanopineoplus2 librespot[963]:                        at src/libstd/sys_common/backtrace.rs:65
May 04 02:26:33 nanopineoplus2 librespot[963]:    7:  0x10b3954 - std::sys_common::backtrace::print::hec570bc9fdd6a536
May 04 02:26:33 nanopineoplus2 librespot[963]:                        at src/libstd/sys_common/backtrace.rs:50
May 04 02:26:33 nanopineoplus2 librespot[963]:    8:  0x10b3954 - std::panicking::default_hook::{{closure}}::hf187be78db8e291b
May 04 02:26:33 nanopineoplus2 librespot[963]:                        at src/libstd/panicking.rs:193
May 04 02:26:33 nanopineoplus2 librespot[963]:    9:  0x10b35b4 - std::panicking::default_hook::h9634e45fc0d61ba6
May 04 02:26:33 nanopineoplus2 librespot[963]:                        at src/libstd/panicking.rs:210
May 04 02:26:33 nanopineoplus2 librespot[963]:   10:  0x10b4030 - std::panicking::rust_panic_with_hook::h35923fdea5783220
May 04 02:26:33 nanopineoplus2 librespot[963]:                        at src/libstd/panicking.rs:471
May 04 02:26:33 nanopineoplus2 librespot[963]:   11:  0x10b3c48 - rust_begin_unwind
May 04 02:26:33 nanopineoplus2 librespot[963]:                        at src/libstd/panicking.rs:375
May 04 02:26:33 nanopineoplus2 librespot[963]:   12:  0x10cfac0 - core::panicking::panic_fmt::hd74e307162bc16e5
[... cut ...]
May 04 02:26:33 nanopineoplus2 librespot[963]:                        at src/libstd/panicking.rs:270
May 04 02:26:33 nanopineoplus2 librespot[963]:   20:  0x10b4484 - std::panic::catch_unwind::ha972c7f8a413a94c
May 04 02:26:33 nanopineoplus2 librespot[963]:                        at src/libstd/panic.rs:394
May 04 02:26:33 nanopineoplus2 librespot[963]:   21:  0x10b4484 - std::rt::lang_start_internal::hdf6cc497202d2594
May 04 02:26:33 nanopineoplus2 librespot[963]:                        at src/libstd/rt.rs:51

```
Any help will be greatly appreciated!

SNTag, 2021-01-10:
Just adding to the thread that I couldn't find raspotify on my phone unless I connect to it first by my computer. Its a neat solution by the way!

closed 2021-11-24 by JasonLG1979