dtcooper/raspotify #181: Raspotify cannot select new playlist or song after one has been selected.

biochemguy, 2019-02-20:
Hi, Raspotify was working well until a week ago. Not sure if something changed on Spotify's end, but I can play a playlist through the spotify app on Raspotify. I can still play/pause and select next previous. However, I cannot choose a new playlist or song once it is playing. On the debug window, It reloads the current song playing. See below:

> pi@hassbian:~ $ sudo systemctl status raspotify
> ● raspotify.service - Raspotify
>    Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
>    Active: active (running) since Tue 2019-02-19 22:00:37 AST; 56s ago
>   Process: 22007 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
>   Process: 22005 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
>  Main PID: 22010 (librespot)
>    CGroup: /system.slice/raspotify.service
>            └─22010 /usr/bin/librespot --name Raspotify --backend alsa --bitrate 160 --disable-audio-cache --enable-volume-normalisation --linear-volume --ini
> 
> Feb 19 22:01:06 hassbian librespot[22010]: INFO:librespot_core::session: Connecting to AP "guc3-accesspoint-b-57f0.ap.spotify.com:4070"
> Feb 19 22:01:06 hassbian librespot[22010]: INFO:librespot_core::session: Authenticated as "edited" !
> Feb 19 22:01:06 hassbian librespot[22010]: INFO:librespot_core::session: Country: "CA"
> Feb 19 22:01:06 hassbian librespot[22010]: INFO:librespot_playback::audio_backend::alsa: Using alsa sink
> Feb 19 22:01:07 hassbian librespot[22010]: INFO:librespot_playback::player: Loading track "Top Shelf" with Spotify URI "spotify:track:79W3GWNsQBpYo4mWRQkLyN"
> Feb 19 22:01:08 hassbian librespot[22010]: INFO:librespot_playback::player: Track "Top Shelf" loaded
> Feb 19 22:01:16 hassbian librespot[22010]: INFO:librespot_playback::player: Loading track "Top Shelf" with Spotify URI "spotify:track:79W3GWNsQBpYo4mWRQkLyN"
> Feb 19 22:01:18 hassbian librespot[22010]: INFO:librespot_playback::player: Track "Top Shelf" loaded
> Feb 19 22:01:22 hassbian librespot[22010]: INFO:librespot_playback::player: Loading track "Top Shelf" with Spotify URI "spotify:track:79W3GWNsQBpYo4mWRQkLyN"
> Feb 19 22:01:24 hassbian librespot[22010]: INFO:librespot_playback::player: Track "Top Shelf" loaded

I was running the latest version and I tried running it without login and also a clean install.

xMordax, 2019-02-20:
Same here, after selecting a song from a I am not able to switch to another playlist

qbovv, 2019-02-20:
Same here. Trying to creative till it's solved, Idunno, but still happy with raspotify.

fl3x21, 2019-02-20:
Hi ! Same here on fresh install (raspbian stretch full).
Simple workaround consist to came back on your device before to select a new song/playlist and then get back on raspotify (really not sure of my english, so I hope it's clear...).
Not the best UX (especially if you switch a lot) but it work well while waiting for issue fixing.
Seem evident but it was not for me during the last 2 days, so... maybe it could help somebody else :-)
By the way, I take the opportunity say a big "Thank You" to the dev - great job on raspotify !

xMordax, 2019-02-20:
as @fl3x21 said, the workaround for now is to switch from Spotify connect to your local Spotify switch playlist and connect back to the Spotify connect

klemment, 2019-02-20:
link to a librespot issue : https://github.com/librespot-org/librespot/issues/288

brunoj, 2019-02-20:
Here as well.

not-william, 2019-02-23:
Here too.

peterdk, 2019-02-23:
This is fixed now, maybe add a new build with the updated librespot?
I compiled myself and it works again!

biochemguy, 2019-02-23:
Yup, updating the program worked for me! I just followed the the installation again by doing: curl -sL https://dtcooper.github.io/raspotify/install.sh | sh

peterdk, 2019-02-23:
@biochemguy maybe keep this issue open until a new build is published for raspotify deb?

biochemguy, 2019-02-23:
This was the same issue as this one:

https://github.com/dtcooper/raspotify/issues/183

I can reopen it if it needs to for the deb package, if it doesn't use the latest librespot package. I don't know exactly how it's built?

forslund, 2019-02-24:
I believe the package has been updated now so this can be closed.

closed 2019-02-24 by biochemguy