dtcooper/raspotify #182: Spotify Client Not Always Visibile II

mikegilchrist, 2019-02-20:


mikegilchrist, 2019-02-20:
I'm running dietpi OS on a RPi3B.  I like to control spotify from my phone, but often the raspotify client is not visible.  Even though raspotify is not visibile the other computer on the network and my Roku are. 

Here's the process information 

`$ ps aux |grep rasp
raspoti+  1204  0.0  0.5  18224  5408 ?        Ssl  Feb17   0:06 /usr/bin/librespot --name raspotify (kitchen) --backend alsa --bitrate 320 --disable-audio-cache --enable-volume-normalisation --linear-volume --initial-volume=85`

 I saw this previous [issue](https://github.com/dtcooper/raspotify/issues/97) where it was fixed by specifying the username and password, but my wife and  I both use this system and have separate spotify accounts, so that doesn't seem like a good solution.  Any suggestions?

mikegilchrist, 2019-02-20:
Okay, after thinking about this some more, I went and checked my wife's phone and saw raspotify was visible.  So I am guessing this issue stems from the fact that we have separate accounts and if she connects with hers the connection is no longer visible on my account.  Any ideas on a fix?  

Can we run raspotify with two credentials?

Could we run two instances of raspotify each using different names/username credentials? 

Something else?

Chris-A-J, 2019-02-25:
I have exactly the same problem since recent 0.12.0 release. Only shows up when username & password is entered, whereas older versions didn't require this if if used on local network. This unfortunately restricts usage to one Spotify account.

I'm aware that Spotify has been recently making changes to their Connect system in the background, resulting in all sorts of software issues.

Any suggestions for a fix? Otherwise, it may be worth rolling back to previous release for now.

closed 2021-11-24 by github-actions[bot]