dtcooper/raspotify #183: Switching songs does not work using spotify connect

yopeople, 2019-02-21:
Hi,

I'm using raspotify on a raspberry 2b with raspbian.

For selecting the music i want to hear i an using an iphone or windows PC with spotify premium., I select a song and I use the option to play this song on the raspberry pi with raspotify on it.. This worked perfectly, i never had a problem.

But since a few days it is not possible anymore to swith from a playing song to another song on an other playlist. When I try this it stucks for a moment and then it play the same song it was playing. When I want to hear another song i now have to disconnect and play the music on the iphone or Windows PC, then i can switch e.g. to another playlist. After switching the playlist i can connect the streaming client again.

I have checked/resetted/reinstalled nearly everthing but the problem still exists - do you have an idea?
I have also tried Volumio with volspotconnect2, but Volumio has the same problem.
Perhaps spotify changed something in their software so that it no longer works 100% with raspotify and Volumio?

janek, 2019-02-21:
thanks for reporting this, I have the same problem, also since a few days.

one more thing that I noticed, in case it helps: 
play/pause, previous and next buttons work, and the 'timeline' of the song also works. 
So I can click on a moment in a current song to jump to it, and if I do that, the next time I try to pick a new song, it will jump to that moment in the current song instead of playing the new one.

I've never heard of Volumio and don't use it, so it's probably not related to that. I agree it's probably a Spotify update. I reported the issue to them and asked for escalation, but not sure how much that will bring.

mertzi, 2019-02-21:
Same issue here. If you chose playback on your phone or desktop version and then stop and start the song you want to play, and then chose raspotify again it sometimes works, but it can take several minutes before playback even starts. I am certain this is issue is because of some change made by the clowns at Spotify and not Raspotify.

Minecon724, 2019-02-21:
You can disconnect, switch song and reconnect.

smithbro2115, 2019-02-21:
I've got the same problem, urg.

brookshank, 2019-02-21:
Same problem, started the last couple of days

polwag, 2019-02-21:
Same problem

zmpetro, 2019-02-22:
Same problem here. Switching to songs in your current queue works but you can't switch to new songs outside your queue.

ericriggsy, 2019-02-22:
I am also having this problem

dtcooper, 2019-02-22:
Waiting on a workable fix or librespot upstream. Will repackage when that's available. Hold tight, folks!

janek, 2019-02-22:
Sweet, thanks @dtcooper . Spotify actually said they will get back to me on this issue within 24-48h, which is very cool of them - so I hope I'll have something to report soon.

0920429, 2019-02-22:
Has been fixed upstream in librespot now.

xMordax, 2019-02-22:
Do we have to wait for the deb package to update?

dtcooper, 2019-02-23:
@xMordax wait no longer! Upstream librespot fixed the bug, and so did raspotify 0.12.0

closed 2019-02-23 by dtcooper

klemment, 2019-02-23:
Same procedure as for first installation, in order to update ?

janek, 2019-02-23:
@klemment, yes, I just ran the 'easy install', got the following output: 

```The following packages will be upgraded:
  raspotify
1 upgraded, 0 newly installed, 0 to remove and 55 not upgraded.```

..and it works, not even a restart necessary. Very user-friendly, thanks a lot @dtcooper.

qbovv, 2019-02-24:
Good news! Now, auto generated playlists (eg daily mix) update when at 'visible end'.