dtcooper/raspotify #186: Can't change playlist

microfx, 2019-03-15:
Hey!

Since a few weeks I cannot change playlists. When I do the current song stops / plays from the beginning but it is not switching to the new playlist. To be able to do that I have to unconnect spotify connect, change playlist and connect again. I still can add songs to queue and that works. Is this a known issue? 

Cheers!

mkoechler, 2019-03-15:
I have the exact same issue. It doesnt matter if I control the raspotify from an Android Phone or Windows PC. Just like microfx the only way to change a song is to disconnect, change the song on the local device and then connect again.

microfx, 2019-03-15:
iOS 12.1 + macOS here

mrpete69, 2019-03-23:
Same here. Android and iOS. Raspotify running on a central RPi 3 with three snapcast clients.

jimsan72, 2019-03-24:
Had the same issue, just updated the Raspberry Pi, worked fine after that.
sudo apt-get update
sudo apt-get dist-upgrade

creisey, 2019-03-25:
thanks a lot, now everything works as before!

krzysiacz, 2019-03-26:
Same here. Update cured the problem. Thanks !

closed 2021-11-24 by github-actions[bot]