dtcooper/raspotify #189: hdmi

Jordopassporto, 2019-03-20:


Jordopassporto, 2019-03-20:
Hello,

I am having an HDMI issue I have gone an extensive amount of forums regarding this issue and have done all of the instructions that were given and still have not been able to hear sound through my HDMI connection. I have the ALSA mixer setup with the correct card number and device.

Please let me know what other things I might be able to do.

Cr4z33, 2019-03-28:
> Hello,
>
> I am having an HDMI issue I have gone an extensive amount of forums regarding this issue and have done all of the instructions that were given and still have not been able to hear sound [...]
I was struggling at understanding what was wrong.

I have entered the correct HDMI output into /etc/default/raspotify and I have forced the HDMI output into /boot/config.txt.

Well it came out I didn't have to force HDMI in the Raspotify config...
As soon as I set the 'bcm2835 ALSA' (in my case hw:0,0) device instead of the 'bcm2835 IEC958/HDMI' one and restarted the Raspotify service the sound started coming out through HDMI!

Jordopassporto, 2019-03-28:
How did you restart the raspotify service?

Cr4z33, 2019-03-28:
sudo systemctl restart raspotify

Also if you want to check its status:

sudo systemctl status raspotify

Jordopassporto, 2019-03-28:
When I put sudo systemctl restart raspotify
I receive this:
Failed to restart raspotify.service: Unit raspotify.service is masked.

So I tried to check the status using sudo systemctl status raspotify. And receive this:
● raspotify.service
   Loaded: masked (/dev/null; bad)
   Active: inactive (dead)

Cr4z33, 2019-03-28:
Hmm not that I am a Linux God, but... was it actually running before?

If yes try rebooting and check again Raspotify service status.

Jordopassporto, 2019-03-28:
yeah it was connected and everything but the hdmi wasnt working for some reason so I  have been playing around with it alot and feel like Im always missing something.

closed 2021-11-24 by github-actions[bot]