dtcooper/raspotify #190: It takes a few minutes after boot before Music is actually played back

Bauspartakus, 2019-03-20:
I just setup my Pi Zero W with a new HiFiBerry DAC+ . I installed Raspbian Stretch Lite and setup Raspotify as described in the readme. 
I plugged the Pi into the AC-socket on the back of my main amplifier, so that it turns on when the amp is turned on.
After turning on the Pi it takes about 20-30 seconds until it shows up in the Device list in my Spotify App, which is perfectly fine.
But when I start playing a song it shows its playing on my App, but there is no sound. When the end of the song is reached the app does not forward to the next song in the playlist.
After a few minutes the song starts playing and everything works as it is supposed to.

Does someone happen to know, what the cause for that could be? Some part of the whole setup taking that long to boot? 

I know, i could just leave the Pi running all the time, but that just doesn't satisfy me :-)

Thanks for your help!

l3d00m, 2019-03-25:
This should be fixed with the next update, especially the not playing after first connect and needing a long time for loading the songs is fixed (https://github.com/librespot-org/librespot)

Semphora, 2019-10-07:
any news on that?

Bauspartakus, 2019-10-12:
Will check later & let you know.

closed 2021-11-24 by github-actions[bot]