dtcooper/raspotify #191: Spotify stops playing after few songs

Csierra099, 2019-03-21:
Hey guys! I've been having some problems with raspotify, we have 3 pi's connected in a hotel(I'm the manager of the IT deparmente) and been using raspotify for about 1 week already in the hotel but since we have this devices with raspotify after some few songs the pi just doesn't send any music signal at all and this happens with all 3 devices, it can play without a problem for at least 3 o 4 hours but we need them to work all day straight with no problem, can someone help me out please! I have my boss quite mad about it and need a solution or alternative.

Thanks!

l3d00m, 2019-03-25:
This is usually the wrong repository to open the issue in, check out https://github.com/librespot-org/librespot/issues and check if someone else has the same issue as you. If you open an issue like this it's also helpful to provide logs as otherwise it's speculation.

This build currently hangs a bit behind librespot, they implemented quite a bit in the last days (@dtcooper :) )

lucianocf, 2019-04-02:
Hey guys! I'm with the same issue. And I've tried to build the package myself with the latest commit of librespot and the build fails. 

@dtcooper , could you please make a new build with the latest version of librespot, please?

kingma1999, 2019-04-15:
removing pulseaudio with
`sudo apt-get remove pulseaudio`
seems to have done the job for me, as suggested in another issue, but I can not find it again, sorry for that.

ljubadr, 2019-07-12:
While I didn't test the @kingma1999 solution, for me the issue was with playing the audio over HDMI cable

In headless mode by default, HDMI signal is turned off after 600 seconds, thus the audio sink gets lost as well.

I've posted my solution in [#112](https://github.com/dtcooper/raspotify/issues/112#issuecomment-510723980)

BUBUsnet, 2019-10-31:
> `sudo apt-get remove pulseaudio`
> seems to have done the job for me

I don't have installed pulseaudio and currently have the same problem that spotify stop after a few songs.

> ERROR:librespot_core::session: Os { code: 104, kind: ConnectionReset, message: "Connection reset by peer"

closed 2021-11-17 by JasonLG1979