dtcooper/raspotify #192: Failed to build librespot, failing at alsa-sys

l3d00m, 2019-03-29:
When I tried to compile using `sudo make`, it first complained about a missing environment variable. After adding `ENV PKG_CONFIG_ALLOW_CROSS 1` to the Dockerfile it now fails with:

```
```

l3d00m, 2019-03-29:
Above commit doesn't work yet

lucianocf, 2019-04-02:
I think I'm with the same issue, but for also-lib:

`configure: error: in `/toolchain/alsa-lib-1.0.29':
configure: error: C compiler cannot create executables
See `config.log' for more details
The command '/bin/sh -c curl -O https://www.mirrorservice.org/sites/ftp.alsa-project.org/pub/lib/alsa-lib-1.0.29.tar.bz2     && tar xvjf alsa-lib-1.0.29.tar.bz2 && cd alsa-lib-1.0.29     && CC=arm-linux-gnueabihf-gcc ./configure --host=arm-linux-gnueabihf --disable-python         --prefix=/toolchain/rpi-tools/arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi/arm-bcm2708hardfp-linux-gnueabi/sysroot     && make -j $(nproc --all) && make install     && cd .. && rm -rf alsa-lib-1.0.29.tar.bz2 alsa-lib-1.0.29' returned a non-zero code: 77
Makefile:9: recipe for target 'raspotify_*.deb' failed
make: *** [raspotify_*.deb] Error 77`

lucianocf, 2019-04-02:
It seems a 64-bit execution error:

alsa-lib-1.0.29/config.log:

`
configure:3438: arm-linux-gnueabihf-gcc  -D_GNU_SOURCE   conftest.c  >&5
./configure: line 3440: /toolchain/rpi-tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/arm-linux-gnueabihf-gcc: cannot execute binary file: Exec format error
configure:3442: $? = 126
configure:3480: result: no
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME "alsa-lib"
| #define PACKAGE_TARNAME "alsa-lib"
| #define PACKAGE_VERSION "1.0.29"
| #define PACKAGE_STRING "alsa-lib 1.0.29"
| #define PACKAGE_BUGREPORT ""
| #define PACKAGE_URL ""
| #define PACKAGE "alsa-lib"
| #define VERSION "1.0.29"
| /* end confdefs.h.  */
| 
| int
| main ()
| {
| 
|   ;
|   return 0;
| }
configure:3485: error: in `/toolchain/alsa-lib-1.0.29':
configure:3487: error: C compiler cannot create executables
`

dtcooper, 2019-04-03:
Noted! Thanks for filing. I'll try to work on a PR, but as an interim solution I can compile and package on my Pi if upstream librespot has necessary changes.

xMordax, 2019-04-05:
Same here when I try to build it:
```
error: failed to run custom build command for `alsa-sys v0.1.2`
process didn't exit successfully: `/build/release/build/alsa-sys-955037715e9ddf22/build-script-build` (exit code: 101)
--- stderr
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: "Cross compilation detected. Use PKG_CONFIG_ALLOW_CROSS=1 to override"', src/libcore/result.rs:997:5
note: Run with `RUST_BACKTRACE=1` environment variable to display a backtrace.

warning: build failed, waiting for other jobs to finish...
error: build failed
Makefile:9: recipe for target 'raspotify_*.deb' failed
make: *** [raspotify_*.deb] Error 101
```

l3d00m, 2019-04-09:
That would be great @dtcooper. They added a lot of stuff recently

lucianocf, 2019-04-11:
@l3d00m I've tried to build on another machine, with Ubuntu x64, added `ENV PKG_CONFIG_ALLOW_CROSS 1` on Dockerfile, but it returned the same message:

> error: failed to run custom build command for `alsa-sys v0.1.2`
> process didn't exit successfully: `/build/release/build/alsa-sys-75b0772c7c28975c/build-script-build` (exit code: 101)
> --- stderr
> thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: "Failed to run `\"pkg-config\" \"--libs\" \"--cflags\" \"alsa\"`: No such file or directory (os error 2)"', src/libcore/result.rs:997:5
> note: Run with `RUST_BACKTRACE=1` environment variable to display a backtrace.
> 
> warning: build failed, waiting for other jobs to finish...
> error: build failed
> Makefile:9: recipe for target 'raspotify_*.deb' failed
> make: *** [raspotify_*.deb] Error 101

lucianocf, 2019-04-11:
> Noted! Thanks for filing. I'll try to work on a PR, but as an interim solution I can compile and package on my Pi if upstream librespot has necessary changes.

@dtcooper , did you had a chance to compile and package with the newer version of librespot? Thank you.

xMordax, 2019-04-12:
Can we get a fix on this?

lucianocf, 2019-04-12:
Tried to build with your "patch-1" branch @l3d00m , but having a fail message while compiling librespot:

```
"
  = note: /toolchain/rpi-tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/../lib/gcc/arm-linux-gnueabihf/4.8.3/../../../../arm-linux-gnueabihf/bin/ld: skipping incompatible /usr/lib/x86_64-linux-gnu/libc.so when searching for -lc
          /usr/lib/x86_64-linux-gnu/libc.a: error adding symbols: File format not recognized
          collect2: error: ld returned 1 exit status
          

error: aborting due to previous error

error: Could not compile `librespot`.

To learn more, run the command again with --verbose.
Makefile:9: recipe for target 'raspotify_*.deb' failed
make: *** [raspotify_*.deb] Error 101

```

dtcooper, 2019-04-12:
I'm going to go ahead and attempt to build this on my actual pi, side stepping the cross-compiling errors. Could take while but expect an update EOD or tomorrow morning PDT. Thanks all!

l3d00m, 2019-04-13:
@dtcooper No worries, thanks for your work :)

dtcooper, 2019-04-15:
Hey all! I have addressed the issue, and cross compiling with `alsa-sys` is now working again. I've tagged version 0.13.0, and you can download the `.deb` file here: [`raspotify_0.13.0.librespot.20190412T034352Z.6c4311f_armhf.deb`](https://github.com/dtcooper/raspotify/releases/download/0.13.0/raspotify_0.13.0.librespot.20190412T034352Z.6c4311f_armhf.deb), which packages [`6c4311f`](https://github.com/librespot-org/librespot/commit/6c4311fe728949aaac49821ed76ef646b146c6a2) of [librespot](https://github.com/librespot-org/librespot). Please test.

I won't have access to a Pi for several more hours, I'll test myself and if it works, I'll deploy to the apt repository then. Thanks for your patience.

closed 2019-04-15 by dtcooper

lucianocf, 2019-04-16:
@dtcooper I have installed on my Pi.. Until now it has played all the songs without stopping, and it's always visible. Thank you so much!

Jiggz0r, 2020-12-23:
Hi,  I'm bumping life into this old thread...

I've got problems compiling this on a RPi 3b, 64-bit Ubuntu. I'm still getting the cross-compiling error:

checking whether the C compiler works... no
configure: error: in `/toolchain/alsa-lib-1.0.29':
configure: error: C compiler cannot create executables
See `config.log' for more details
The command '/bin/sh -c curl -O https://www.mirrorservice.org/sites/ftp.alsa-project.org/pub/lib/alsa-lib-1.0.29.tar.bz2     && tar xvjf alsa-lib-1.0.29.tar.bz2 && cd alsa-lib-1.0.29     && CC=arm-linux-gnueabihf-gcc ./configure --host=arm-linux-gnueabihf --disable-python         --prefix=/toolchain/rpi-tools/arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi/arm-bcm2708hardfp-linux-gnueabi/sysroot     && make -j $(nproc --all) && make install     && cd .. && rm -rf alsa-lib-1.0.29.tar.bz2 alsa-lib-1.0.29' returned a non-zero code: 77
make: *** [Makefile:9: raspotify_*.deb] Error 77

I had a look at the Dockerfile, and there already seem to be a cross-compilation directive:

# Install 1.0.x alsa-utils which is needed for compilation
ENV PKG_CONFIG_ALLOW_CROSS 1

However, when I compile, this seems to be ignored. Any ideas?

(Using 64bit Ubuntu because most of the containers I'm using are built for arm64).

Cheers,
JS

commit 8fcfb6e0 (2019-04-15): Fixed cross-compiling of alsa-sys (closes #192)