dtcooper/raspotify #193: Spotify stops playing: Error sending packet OS

Csierra099, 2019-04-01:
Hey guys! So after using Raspotify for about 3 weeks already I have encountred an issue that has to do with #112, issue that I posted but didnt really know what it was, after seeing what the log said I encounted with a problem that says Error Sending Packet OS, anyone knows what I could be?

pi@MediaCenter:~ $ journalctl --system _SYSTEMD_UNIT=raspotify.service
-- Logs begin at Mon 2019-04-01 12:46:55 EDT, end at Mon 2019-04-01 16:15:42 EDT. --
Apr 01 12:47:08 MediaCenter librespot[435]: INFO:librespot: librespot (raspotify v0.12.0) ef28df6 (2019-03-14). Built on 2019-03-14. Build ID: BFEig1gv
Apr 01 12:47:09 MediaCenter librespot[435]: WARN:<unknown>: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
Apr 01 12:47:09 MediaCenter librespot[435]: INFO:librespot_core::session: Connecting to AP "guc3-accesspoint-b-467m.ap.spotify.com:4070"
Apr 01 12:47:09 MediaCenter librespot[435]: INFO:librespot_core::session: Authenticated as "hotelabyt" !
Apr 01 12:47:09 MediaCenter librespot[435]: INFO:librespot_playback::audio_backend::alsa: Using alsa sink
Apr 01 12:47:09 MediaCenter librespot[435]: INFO:librespot_core::session: Country: "MX"
Apr 01 13:00:02 MediaCenter librespot[435]: INFO:librespot_playback::player: Loading track "The Joke" with Spotify URI "spotify:track:37nD8NJhod8BEZ6EGYo2gT"
Apr 01 13:00:04 MediaCenter librespot[435]: INFO:librespot_playback::player: Track "The Joke" loaded
Apr 01 13:02:20 MediaCenter librespot[435]: INFO:librespot_playback::player: Loading track "Is That Alright?" with Spotify URI "spotify:track:172e82oK9vGoBPvc6SqMm5"
Apr 01 13:02:21 MediaCenter librespot[435]: INFO:librespot_playback::player: Track "Is That Alright?" loaded
Apr 01 13:05:32 MediaCenter librespot[435]: INFO:librespot_playback::player: Loading track "Hold Your Head Up High - Recorded at Spotify Studios NYC" with Spotify URI "
Apr 01 13:05:32 MediaCenter librespot[435]: INFO:librespot_playback::player: Track "Hold Your Head Up High - Recorded at Spotify Studios NYC" loaded
Apr 01 13:09:06 MediaCenter librespot[435]: INFO:librespot_playback::player: Loading track "No One's Gonna Love You (Stockholm Version)" with Spotify URI "spotify:track
Apr 01 13:09:06 MediaCenter librespot[435]: INFO:librespot_playback::player: Track "No One's Gonna Love You (Stockholm Version)" loaded
Apr 01 13:12:55 MediaCenter librespot[435]: INFO:librespot_playback::player: Loading track "Sign of the Times" with Spotify URI "spotify:track:5ELRkzdzz0HvGpMDlfZHkV"
Apr 01 13:12:55 MediaCenter librespot[435]: INFO:librespot_playback::player: Track "Sign of the Times" loaded
Apr 01 13:18:35 MediaCenter librespot[435]: INFO:librespot_playback::player: Loading track "If You Ever Wanna Be In Love - James Bay Spotify Session 2015" with Spotify
Apr 01 13:18:35 MediaCenter librespot[435]: INFO:librespot_playback::player: Track "If You Ever Wanna Be In Love - James Bay Spotify Session 2015" loaded
Apr 01 13:22:32 MediaCenter librespot[435]: INFO:librespot_playback::player: Loading track "Lonesome" with Spotify URI "spotify:track:0j3v2MeAl14a7I1F3vvewM"
Apr 01 13:22:32 MediaCenter librespot[435]: INFO:librespot_playback::player: Track "Lonesome" loaded
Apr 01 13:26:22 MediaCenter librespot[435]: INFO:librespot_playback::player: Loading track "Waking Up Slow - Piano" with Spotify URI "spotify:track:7E9QA6hL8Ebk0yHXeV1U
Apr 01 13:26:22 MediaCenter librespot[435]: INFO:librespot_playback::player: Track "Waking up Slow - Piano Version" loaded
Apr 01 13:30:03 MediaCenter librespot[435]: INFO:librespot_playback::player: Loading track "Remedy" with Spotify URI "spotify:track:7GgQi7JTG4b6J4iEF4RTjF"
Apr 01 13:30:03 MediaCenter librespot[435]: INFO:librespot_playback::player: Track "Remedy" loaded
Apr 01 13:34:09 MediaCenter librespot[435]: INFO:librespot_playback::player: Loading track "Don't Let Me Down" with Spotify URI "spotify:track:6oqlWkAUCqBXjvorIRSm54"
Apr 01 13:34:09 MediaCenter librespot[435]: INFO:librespot_playback::player: Track "Don't Let Me Down" loaded
Apr 01 13:37:09 MediaCenter librespot[435]: INFO:librespot_playback::player: Loading track "What A Wonderful World" with Spotify URI "spotify:track:2a2gJ4YinLgIovY75zA0
Apr 01 13:37:09 MediaCenter librespot[435]: INFO:librespot_playback::player: Track "What A Wonderful World" loaded
Apr 01 13:42:02 MediaCenter librespot[435]: INFO:librespot_playback::player: Loading track "I Get to Love You" with Spotify URI "spotify:track:3CV4lFELdziNM1OnHU8nwz"
Apr 01 13:42:03 MediaCenter librespot[435]: INFO:librespot_playback::player: Track "I Get to Love You" loaded
Apr 01 13:46:02 MediaCenter librespot[435]: INFO:librespot_playback::player: Loading track "The Best You Had - Acoustic Version" with Spotify URI "spotify:track:1JyMina
Apr 01 13:46:02 MediaCenter librespot[435]: INFO:librespot_playback::player: Track "The Best You Had - Acoustic Version" loaded
**Apr 01 13:46:36 MediaCenter librespot[435]: WARN:<unknown>: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
Apr 01 13:46:38 MediaCenter librespot[435]: WARN:<unknown>: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
Apr 01 13:46:45 MediaCenter librespot[435]: WARN:<unknown>: error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }**
Apr 01 13:49:20 MediaCenter librespot[435]: INFO:librespot_playback::player: Loading track "Too Good At Goodbyes" with Spotify URI "spotify:track:3VlbOrM6nYPprVvzBZllE5
Apr 01 13:49:20 MediaCenter librespot[435]: INFO:librespot_playback::player: Track "Too Good At Goodbyes" loaded
Apr 01 13:52:41 MediaCenter librespot[435]: INFO:librespot_playback::player: Loading track "Make You Feel My Love" with Spotify URI "spotify:track:7rPLZ8Krm6CZIbraFUlnW
Apr 01 13:52:41 MediaCenter librespot[435]: INFO:librespot_playback::player: Track "Make You Feel My Love" loaded
Apr 01 13:56:13 MediaCenter librespot[435]: INFO:librespot_playback::player: Loading track "A Thousand Years" with Spotify URI "spotify:track:6lanRgr6wXibZr8KgzXxBl"
Apr 01 13:56:13 MediaCenter librespot[435]: INFO:librespot_playback::player: Track "A Thousand Years" loaded
Apr 01 14:00:58 MediaCenter librespot[435]: INFO:librespot_playback::player: Loading track "All On My Mind - Acoustic" with Spotify URI "spotify:track:4piJ8pnoRvC5sqMEp
Apr 01 14:00:58 MediaCenter librespot[435]: INFO:librespot_playback::player: Track "All On My Mind - Acoustic" loaded
Apr 01 14:04:47 MediaCenter librespot[435]: INFO:librespot_playback::player: Loading track "Somewhere Only We Know" with Spotify URI "spotify:track:0ll8uFnc0nANY35E0Lfx
Apr 01 14:04:48 MediaCenter librespot[435]: INFO:librespot_playback::player: Track "Somewhere Only We Know" loaded
Apr 01 14:08:44 MediaCenter librespot[435]: INFO:librespot_playback::player: Loading track "Stitches" with Spotify URI "spotify:track:3Jy394MtpM2xciRBTzxBlg"
Apr 01 14:08:44 MediaCenter librespot[435]: INFO:librespot_playback::player: Track "Stitches" loaded
Apr 01 14:12:50 MediaCenter librespot[435]: INFO:librespot_playback::player: Loading track "In a New Bed" with Spotify URI "spotify:track:4Or1Yp4i3VANJE6E39ys91"
Apr 01 14:12:50 MediaCenter librespot[435]: INFO:librespot_playback::player: Track "In a New Bed" loaded
Apr 01 14:16:31 MediaCenter librespot[435]: INFO:librespot_playback::player: Loading track "Helium" with Spotify URI "spotify:track:4S6fv0puLCsfYjyBTPDb9k"
Apr 01 14:16:31 MediaCenter librespot[435]: INFO:librespot_playback::player: Track "Helium" loaded
Apr 01 14:20:43 MediaCenter librespot[435]: INFO:librespot_playback::player: Loading track "The Scientist" with Spotify URI "spotify:track:75JFxkI2RXiU7L9VXzMkle"
Apr 01 14:20:44 MediaCenter librespot[435]: INFO:librespot_playback::player: Track "The Scientist" loaded
Apr 01 14:25:53 MediaCenter librespot[435]: INFO:librespot_playback::player: Loading track "Stone" with Spotify URI "spotify:track:6cRExfWfj87zhACU30aW7l"
Apr 01 14:25:53 MediaCenter librespot[435]: INFO:librespot_playback::player: Track "Stone" loaded
Apr 01 14:29:29 MediaCenter librespot[435]: INFO:librespot_playback::player: Loading track "hate u love u" with Spotify URI "spotify:track:6ATgDc6e4sPn84hvJsAmPt"
Apr 01 14:29:30 MediaCenter librespot[435]: INFO:librespot_playback::player: Track "hate u love u" loaded
[... cut ...]
Apr 01 16:11:11 MediaCenter librespot[435]: INFO:librespot_playback::player: Loading track "Shelter" with Spotify URI "spotify:track:1sLxooNd8EKRLDs0YiwbZF"
Apr 01 16:11:11 MediaCenter librespot[435]: INFO:librespot_playback::player: Track "Shelter" loaded
Apr 01 16:15:41 MediaCenter librespot[435]: INFO:librespot_playback::player: Loading track "Blackout" with Spotify URI "spotify:track:0bqQuzIX7tpY4IBllPHDqF"
Apr 01 16:15:42 MediaCenter librespot[435]: INFO:librespot_playback::player: Track "Blackout" loaded

dthisner, 2019-04-20:
I have the same issue

dtcooper, 2019-04-20:
Might you try the most recent version?

agneevX, 2020-08-08:
Same issue here

```log
Aug 08 00:40:45 pi4 librespot[911]: [2020-08-07T19:10:45Z WARN  libmdns::fsm] error sending packet Os { code: 99, kind: AddrNotAvailable, message: "Cannot assign requested address" }
```

greyivy, 2020-12-13:
Same issue.

JasonLG1979, 2021-11-24:
This is clearly and upstream issue. I would update to the latest version of Raspotify and if the issue persists file a bug report upstream with librespot.

closed 2021-11-24 by JasonLG1979