dtcooper/raspotify #194: Raspotify Slow Song

kolone13, 2019-04-02:
Hey,  i User osmc and installed Raspotify. IT worked but the music Sound is slower and deeper.  It's like when i play the song in half speed

jeremymatthewwerner, 2019-04-28:
I'm having the same problem.
On stretch.

jeremymatthewwerner, 2019-04-29:
I found a fix that worked for me here: https://askubuntu.com/a/1051160

Copy-and-pasting it here:

---
Press Ctrl+Alt+T to go to a terminal and use your favourite editor to edit the file

nano /etc/pulse/default.pa
then find a line containing:

load-module module-udev-detect
modify this to become:

load-module module-udev-detect tsched=0
save and exit and you're all set!
---

kolone13, 2019-04-30:
Thank you very much!I tried it on stretch and it also worked for me. On osmc i didn't  had the problem after i enabled the airplay. So it was not nescessary to install raspotify

closed 2019-04-30 by kolone13