dtcooper/raspotify #197: No sound

TooMuchRAM, 2019-04-10:
Exactly what the title says. Fresh Raspbian install, fresh Raspotify install. 
`omxplayer example.mp3 -o local` works just fine.
I had it running before, but at some point the music became distorted, and after a reboot it stopped playing at all. I then reimaged my Pi. Only other two things that are running are dnsmasq and hostapd. Tried PulseAudio with no luck.

Please help.

gonzalo, 2019-04-11:
Today I had a similar issue, working raspotiy stop working from one day to next (seems working fine but no sound). Reinstall raspotify fixed the issue.

TooMuchRAM, 2019-04-13:
I tried reinstalling it, and reinstalling Raspbian, but it did nothing. Still no audio.

peterdk, 2019-05-18:
Since a week I am having issues that raspotify won't play anything, while the progressbar in Spotify just continues. I tried recompile with newest librespot, but it's not working. Also tried restart etc.

dtcooper, 2019-05-19:
Will look at this today. Thanks for the report.

peterdk, 2019-05-19:
For me it seems to be the same songs on a playlist, the last few. Everytime it reaches there, it will stop and I need to restart Raspotify.

ianshepherd, 2019-05-27:
I might be having a similar issue - I still have sound, but it could be described as "distorted", as OP says. It's actually like it's inserting a large number of tiny gaps per second into the audio, so the song is playing slowly because it's stretched out by the gaps.

Tried apt update, reinstalling package, reducing/increasing memory split - no effect. omxplayer example.mp3 works fine

andrewtvuong, 2019-06-01:
try command 'pkill pulseaudio'

ovidiumoraru, 2019-11-09:
I tried to kill pulseaudio and then restarted it. I change the audio source to raspotify and it started playing, however it stopped after 2-3 seconds.

ovidiumoraru, 2019-11-09:
I eventually removed pulseaudio, rebooted and everything works like a charm.

worldsayshi, 2021-09-13:
I also had some issues after trying raspotify after long time of not using it. Not sure what exactly fixed it at this point. Tried updating and reinstalling but that didn't seem to help. It seemed to run. Spotify connected with no issue. `speaker-test` worked. No errors in `systemctl status raspotify`.

One of these things probably helped:
- Restart device without hdmi cable (My speakers are using analog aoutput. My Hdmi device didn't make any noise previous to this)
- Restart raspotify with sudo: `sudo systemctl restart raspotify`

worldsayshi, 2021-09-13:
It seems that my problem was hdmi and 3.5-mm headphone jack output fighting it out. It looks like I found a permanent solution by setting headphone jack as default device:

In `/etc/asound.conf`:
````
pcm.!default {
    type hw
    card 1
}

ctl.!default {
    type hw           
    card 1
}
````

From: https://superuser.com/a/630048/117486

closed 2021-11-24 by JasonLG1979