dtcooper/raspotify #201: Connection reset by peer

lucianocf, 2019-04-22:
Hello guys. I need some help.

I have updated the Raspotify package to the latest available. Today after playing about 2 songs, the Spotify disconnect on the middle of the second song, and after the song ends, the Raspotify still visible, but I can't connect with any device on my network. 

Here is the logs:
```
Apr 21 23:53:52 black-pearl systemd[1]: Starting Raspotify...
Apr 21 23:53:52 black-pearl systemd[1]: Started Raspotify.
Apr 21 23:53:52 black-pearl librespot[1129]: INFO:librespot: librespot (raspotify v0.13.0) 5d2cb32 (2019-04-16). Built on 2019-04-19. Build ID: ygiDF0Qz
Apr 21 23:55:17 black-pearl librespot[1129]: INFO:librespot_core::session: Connecting to AP "guc3-accesspoint-b-t53h.ap.spotify.com:4070"
Apr 21 23:55:18 black-pearl librespot[1129]: INFO:librespot_core::session: Authenticated as "lucianocf" !
Apr 21 23:55:18 black-pearl librespot[1129]: INFO:librespot_playback::audio_backend::alsa: Using alsa sink
Apr 21 23:55:18 black-pearl librespot[1129]: INFO:librespot_core::session: Country: "BR"
Apr 21 23:55:19 black-pearl librespot[1129]: INFO:librespot_playback::player: Loading track "Times Like These - Santa Barbara, CA" with Spotify URI "spotify:track:41K4TYk61rnnZIVMoX33SU"
Apr 21 23:55:24 black-pearl librespot[1129]: INFO:librespot_playback::player: Track "Times Like These - Santa Barbara, CA" loaded
Apr 21 23:55:25 black-pearl librespot[1129]: INFO:librespot_playback::player: Loading track "Times Like These - Santa Barbara, CA" with Spotify URI "spotify:track:41K4TYk61rnnZIVMoX33SU"
Apr 21 23:55:25 black-pearl librespot[1129]: INFO:librespot_playback::player: Track "Times Like These - Santa Barbara, CA" loaded
Apr 21 23:57:46 black-pearl librespot[1129]: INFO:librespot_playback::player: Loading track "Angel / Better Together - Paris, France" with Spotify URI "spotify:track:0duNRDsScYKeF6Zl3fEhF
Apr 21 23:57:46 black-pearl librespot[1129]: INFO:librespot_playback::player: Track "Angel / Better Together - Paris, France" loaded
Apr 22 00:04:00 black-pearl librespot[1129]: ERROR:librespot_core::session: Os { code: 104, kind: ConnectionReset, message: "Connection reset by peer" }
```

bratwurstmuesli, 2019-05-02:
same here:

May 02 08:25:47 raspberrypi librespot[16267]: INFO:librespot_playback::audio_backend::alsa: Using alsa sink
May 02 08:25:50 raspberrypi librespot[16267]: INFO:librespot_playback::player: Loading track "Wo niemals Ebbe ist" with Spotify URI "spotify:track:1wf7HJ9eTLCyNy1COa3xj3"
May 02 08:25:52 raspberrypi librespot[16267]: INFO:librespot_playback::player: Track "Wo niemals Ebbe ist" loaded
May 02 08:26:05 raspberrypi librespot[16267]: INFO:librespot_playback::player: Loading track "99 Proofs" with Spotify URI "spotify:track:2378GXDstPcQlx59GAggTU"
May 02 08:26:05 raspberrypi librespot[16267]: INFO:librespot_playback::player: Track "99 Proofs" loaded
May 02 08:28:44 raspberrypi librespot[16267]: INFO:librespot_playback::player: Loading track "Storm" with Spotify URI "spotify:track:6aSBwHHB2w4TJsgiZluyEU"
May 02 08:28:45 raspberrypi librespot[16267]: INFO:librespot_playback::player: Track "Storm" loaded
May 02 08:28:53 raspberrypi librespot[16267]: INFO:librespot_playback::player: Loading track "Wild Stare" with Spotify URI "spotify:track:2SClPoLfPJmwrr3EyfgRTS"
May 02 08:28:53 raspberrypi librespot[16267]: INFO:librespot_playback::player: Track "Wild Stare" loaded
May 02 08:32:08 raspberrypi librespot[16267]: ERROR:librespot_core::session: Os { code: 104, kind: ConnectionReset, message: "Connection reset by peer" }

bratwurstmuesli, 2019-05-02:
i have the same problem! did you find any way to fix this?

michaeljohnbennett, 2019-06-17:
yes, same here, I also find that raspotify can't be seen from desktop apps without a phone app seeing it first.

bit frustrating.

lucianocf, 2019-06-17:
> i have the same problem! did you find any way to fix this?

No, I didn't finda a way yet. Tried to compile with a newer version of librespot, but it fails.

michaeljohnbennett, 2019-06-17:
I have this running on a screenly OSE install and its working again, seems to be very intermittent and linked to a number of issues around different spotify clients on your network. I'm trying it with fixed username/login for the next while to see if that fixes things.

mattfro, 2019-12-30:
Same issue her also... I have upgraded to later librespot from november.

microfx, 2020-01-14:
no fix? that is not cool... spotify connect is the best feature ever...

tomtexp, 2020-08-26:
Is there something new on this error?

bwagley, 2020-12-16:
I was having this same issue. Seemed to be connected to the strength of connection to the pi. This pi was on a 2.4Ghz network that had been running slow for unrelated reasons, and would have this error. When switched onto the 5Ghz network that was having no problems, raspotify began working again.

closed 2021-11-24 by JasonLG1979