dtcooper/raspotify #202: Mac client must connect to Raspotify before Mycroft skill sees it

jeremymatthewwerner, 2019-04-28:
I have installed Raspotify and the Mycroft Spotify Skill on the same Raspberry Pi and I have them connecting to my Premium Spotify account and playing through Raspotify on the same machine - with one catch.

On a fresh startup when I ask Mycroft to play something from Spotify, it can’t find any devices, saying " Spotify couldn’t start playback. I couldn’t find any Spotify devices".

ps -ef | grep reaspot shows /usr/bin/librespot is running, but for some reason the skill can’t find it.

However, If I then launch the Spotify app on my Mac and go to Settings -> Open Devices Menu -> and click the device that shows called “raspotify” it somehow “pokes” Raspotify because immediately re-issuing the same voice command to Mycroft works! I don’t even have to start a stream playing from my Mac first. Just clicking the device on the Mac client seems to make Raspotify visible to Mycroft on the Rasperry Pi both are running on.

Has anyone seen this behavior before? Does anyone have any idea what might be going on?

mwargan, 2019-07-04:
I have similar issues with Raspotify not showing up on my iPhone, but showing up on my Mac. When connecting via Mac, it "pokes" (as you say) the iPhone and it suddenly shows up on the list.

mashers, 2019-08-06:
I am having the same problem. I cannot get Mycroft to recognise the Raspotify device even though I have configured the Mycroft Spotify skill with the correct device name. When I choose the device from the list on my iPhone I can stream to it, and even before I do this the service is running on the Pi.

> $ ps -ef | grep raspot
> raspoti+   972     1  0 16:56 ?        00:00:00 /usr/bin/librespot --name Picroft --backend alsa --bitrate 320 --disable-audio-cache --enable-volume-normalisation --linear-volume --initial-volume=100

closed 2021-11-24 by github-actions[bot]