dtcooper/raspotify #204: Detect if music is playing

fankijo, 2019-05-01:
Hi,

I was wondering if there was any chance for other scripts running on the pi to detect wether Music is beeing played or not. My goal is to run a command as soon as msic is beeing played and run another command as soon as I disconnect my spotify off the raspotify.

olafWal, 2019-05-04:
Add "--onevent /path/to/your/script" to librespot options

e.g.:
---------------8<----------------------
#! /bin/bash

logger $PLAYER_EVENT
logger $TRACK_ID
-------------->8----------------------------

$PLAYER_EVENT can be start|change|stop

20100times, 2020-05-05:
Hi

First thanks for your job, this tool is nice.
Like @fankijo I was wondering if I can run a python script (to turn on my hifi system) as soon as I start playing music on my raspberry pi.
@olafWal seems to have an idea about it but I can't figure his answer out.

Any help is welcome. Thank you for that.

20100times, 2020-05-06:
Hi @fankijo !

Did you manage to run a script when the music is played ?
Thanks

closed 2021-11-24 by github-actions[bot]