dtcooper/raspotify #205: EBUSY: Device or resource busy

onlyyou, 2019-05-11:
Hi all,
Since last update (with OSMC) can't get raspotify to work

Option line should be good
`
osmc@osmc:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: BossDAC [BossDAC], device 0: Boss DAC HiFi [Master] pcm512x-hifi-0 []
  Subdevices: 0/1
  Subdevice #0: subdevice #0
`

`
OPTIONS="--device plughw:0,0"
`

The log

`* raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2019-05-11 08:32:06 CEST; 17s ago
  Process: 13540 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 13537 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 13543 (librespot)
   CGroup: /system.slice/raspotify.service
           `-13543 /usr/bin/librespot --name raspotify (osmc) --backend alsa --bitrate 160 --disable-audio-cache --enable-volume-normalisation --linear-volume --initial-volume=100 --device plughw:0,0

May 11 08:32:19 osmc librespot[13543]: INFO:librespot_core::session: Authenticated as "blogonat" !
May 11 08:32:19 osmc librespot[13543]: INFO:librespot_playback::audio_backend::alsa: Using alsa sink
May 11 08:32:19 osmc librespot[13543]: INFO:librespot_core::session: Country: "FR"
May 11 08:32:19 osmc librespot[13543]: INFO:librespot_playback::player: Loading track "Lover Of The Light" with Spotify URI "spotify:track:73ZdoUAA4qgXrz7j30qirm"
May 11 08:32:20 osmc librespot[13543]: INFO:librespot_connect::spirc: Resolved 20 tracks from <"spotify:station:album:5DWkj0MS3zVHmFXHbjcwf6">
May 11 08:32:20 osmc librespot[13543]: INFO:librespot_playback::player: Track "Lover Of The Light" loaded
May 11 08:32:20 osmc librespot[13543]: ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'EBUSY: Device or resource busy'
May 11 08:32:20 osmc librespot[13543]: ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
May 11 08:32:20 osmc librespot[13543]: ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'EBUSY: Device or resource busy'
May 11 08:32:20 osmc librespot[13543]: ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
`

Many thanks !

Rudolf112, 2019-08-17:
I think I have a very similar issue. Trying to use DragonFly USB DAC, getting either one of two errors:

`osmc@osmc:~$ service raspotify status
* raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2019-08-17 13:49:17 EEST; 21s ago
  Process: 629 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 626 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 631 (librespot)
   CGroup: /system.slice/raspotify.service
           `-631 /usr/bin/librespot --name raspotify --backend alsa --bitrate 320 --disable-audio-cache --enable-volume-normalisation --linear-volume --initial-volume=100 --device hw:0,0

Aug 17 13:49:27 osmc librespot[631]: ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'EBUSY: Device or resource busy'
Aug 17 13:49:27 osmc librespot[631]: ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
Aug 17 13:49:32 osmc librespot[631]: ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'EBUSY: Device or resource busy'
Aug 17 13:49:32 osmc librespot[631]: ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
Aug 17 13:49:36 osmc librespot[631]: INFO:librespot_playback::player: Loading track "It's Been Awhile" with Spotify URI "spotify:track:1BExV8JprNJLvLeDE9qdEW"
Aug 17 13:49:36 osmc librespot[631]: INFO:librespot_playback::player: Track "It's Been Awhile" loaded
Aug 17 13:49:36 osmc librespot[631]: ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'EBUSY: Device or resource busy'
Aug 17 13:49:36 osmc librespot[631]: ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
Aug 17 13:49:36 osmc librespot[631]: ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'EBUSY: Device or resource busy'
Aug 17 13:49:36 osmc librespot[631]: ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
`
Then when I try to restart the raspotify service, I might get:
`osmc@osmc:~$ service raspotify status
* raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2019-08-17 13:50:49 EEST; 678ms ago
  Process: 911 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 909 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 914 (librespot)
   CGroup: /system.slice/raspotify.service
           `-914 /usr/bin/librespot --name raspotify --backend alsa --bitrate 320 --disable-audio-cache --enable-volume-normalisation --linear-volume --initial-volume=100 --device hw:0,0

Aug 17 13:50:49 osmc systemd[1]: Starting Raspotify...
Aug 17 13:50:49 osmc systemd[1]: Started Raspotify.
Aug 17 13:50:49 osmc librespot[914]: INFO:librespot: librespot (raspotify v0.13.1) 5d2cb32 (2019-04-16). Built on 2019-04-29. Build ID: bQrzTj3V
Aug 17 13:50:49 osmc librespot[914]: WARN:<unknown>: Failed to register IPv6 receiver: Os { code: 19, kind: Other, message: "No such device" }
osmc@osmc:~$ service raspotify status
* raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2019-08-17 13:50:49 EEST; 10s ago
  Process: 911 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 909 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 914 (librespot)
   CGroup: /system.slice/raspotify.service
           `-914 /usr/bin/librespot --name raspotify --backend alsa --bitrate 320 --disable-audio-cache --enable-volume-normalisation --linear-volume --initial-volume=100 --device hw:0,0

Aug 17 13:50:56 osmc librespot[914]: INFO:librespot_core::session: Connecting to AP "gew1-accesspoint-a-mmzw.ap.spotify.com:4070"
Aug 17 13:50:56 osmc librespot[914]: INFO:librespot_core::session: Authenticated as "1176469547" !
Aug 17 13:50:56 osmc librespot[914]: INFO:librespot_playback::audio_backend::alsa: Using alsa sink
Aug 17 13:50:56 osmc librespot[914]: INFO:librespot_core::session: Country: "EE"
Aug 17 13:50:56 osmc librespot[914]: INFO:librespot_playback::player: Loading track "Shallow - Radio Edit" with Spotify URI "spotify:track:6QfS2wq5sSC1xAJCQsTSlj"
Aug 17 13:50:58 osmc librespot[914]: INFO:librespot_playback::player: Track "Shallow - Radio Edit" loaded
Aug 17 13:50:58 osmc librespot[914]: ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function 'snd_pcm_hw_params_set_format' failed with error 'EINVAL: Invalid argument'
Aug 17 13:50:58 osmc librespot[914]: ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
Aug 17 13:50:58 osmc librespot[914]: ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function 'snd_pcm_hw_params_set_format' failed with error 'EINVAL: Invalid argument'
Aug 17 13:50:58 osmc librespot[914]: ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
`

Any help is greatly appreciated. Funny thing is that a noname Chinese USB DAC worked just fine and when switching to DragonFly, can't get it to work.

carlosperate, 2020-10-04:
I'm having a similar issue using the Google Voice Kit. The daemon log shows this crash:

```
$ sudo journalctl -u raspotify.service
Oct 04 22:13:43 raspberrypi systemd[1]: /lib/systemd/system/raspotify.service:6: Unknown lvalue 'WantedBy' in section 'Service', ignoring
Oct 04 22:13:44 raspberrypi systemd[1]: Started Raspotify.
Oct 04 22:13:44 raspberrypi librespot[1063]: [2020-10-04T21:13:44Z INFO  librespot] librespot (raspotify v0.14.0) 6aac8a6 (2020-07-24). Built on 2020-07-26. Build ID: wPNyqn0m
Oct 04 22:13:55 raspberrypi librespot[1063]: [2020-10-04T21:13:55Z INFO  librespot_core::session] Connecting to AP "gew1-accesspoint-c-mxmn.ap.spotify.com:4070"
Oct 04 22:13:55 raspberrypi librespot[1063]: [2020-10-04T21:13:55Z INFO  librespot_core::session] Authenticated as "zeroneo" !
Oct 04 22:13:55 raspberrypi librespot[1063]: [2020-10-04T21:13:55Z INFO  librespot_playback::audio_backend::alsa] Using alsa sink
Oct 04 22:13:55 raspberrypi librespot[1063]: [2020-10-04T21:13:55Z INFO  librespot_core::session] Country: "GB"
Oct 04 22:13:56 raspberrypi librespot[1063]: [2020-10-04T21:13:56Z INFO  librespot_playback::player] Loading <Drown> with Spotify URI <spotify:track:6o39Ln9118FKTMbM4BvcEy>
Oct 04 22:13:57 raspberrypi librespot[1063]: [2020-10-04T21:13:57Z INFO  librespot_playback::player] <Drown> (222093 ms) loaded
Oct 04 22:13:57 raspberrypi librespot[1063]: [2020-10-04T21:13:57Z INFO  librespot_playback::player] Loading <Singularity> with Spotify URI <spotify:track:1D2fvbNf4LpOGPPkJ4uqyV>
Oct 04 22:13:57 raspberrypi librespot[1063]: [2020-10-04T21:13:57Z INFO  librespot_playback::player] <Singularity> (172653 ms) loaded
Oct 04 22:14:00 raspberrypi librespot[1063]: [2020-10-04T21:14:00Z ERROR librespot_playback::audio_backend::alsa] Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'EBUSY: Device or resource busy'
Oct 04 22:14:00 raspberrypi librespot[1063]: [2020-10-04T21:14:00Z ERROR librespot_playback::player] Could not start audio: Alsa error: PCM open failed
Oct 04 22:14:00 raspberrypi librespot[1063]: [2020-10-04T21:14:00Z ERROR librespot_playback::audio_backend::alsa] Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'EBUSY: Device or resource busy'
Oct 04 22:14:00 raspberrypi librespot[1063]: [2020-10-04T21:14:00Z ERROR librespot_playback::player] Could not start audio: Alsa error: PCM open failed
Oct 04 22:14:00 raspberrypi librespot[1063]: [2020-10-04T21:14:00Z ERROR librespot_playback::audio_backend::alsa] Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'EBUSY: Device or resource busy'
Oct 04 22:14:00 raspberrypi librespot[1063]: [2020-10-04T21:14:00Z ERROR librespot_playback::player] Could not start audio: Alsa error: PCM open failed
Oct 04 22:14:00 raspberrypi librespot[1063]: [2020-10-04T21:14:00Z ERROR librespot_playback::audio_backend::alsa] Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'EBUSY: Device or resource busy'
Oct 04 22:14:00 raspberrypi librespot[1063]: [2020-10-04T21:14:00Z ERROR librespot_playback::player] Could not start audio: Alsa error: PCM open failed
Oct 04 22:14:00 raspberrypi librespot[1063]: [2020-10-04T21:14:00Z ERROR librespot_playback::audio_backend::alsa] Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'EBUSY: Device or resource busy'
Oct 04 22:14:00 raspberrypi librespot[1063]: [2020-10-04T21:14:00Z ERROR librespot_playback::player] Could not start audio: Alsa error: PCM open failed
Oct 04 22:14:00 raspberrypi librespot[1063]: thread '<unnamed>' panicked at 'called `Option::unwrap()` on a `None` value', playback/src/audio_backend/alsa.rs:103:19
Oct 04 22:14:00 raspberrypi librespot[1063]: stack backtrace:
Oct 04 22:14:01 raspberrypi librespot[1063]:    0:   0x929584 - backtrace::backtrace::libunwind::trace::had1ce870e7642f03
Oct 04 22:14:01 raspberrypi librespot[1063]:                        at /cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.46/src/backtrace/libunwind.rs:86
Oct 04 22:14:01 raspberrypi librespot[1063]:    1:   0x929584 - backtrace::backtrace::trace_unsynchronized::he46c884781dd51a8
Oct 04 22:14:01 raspberrypi librespot[1063]:                        at /cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.46/src/backtrace/mod.rs:66
Oct 04 22:14:01 raspberrypi librespot[1063]:    2:   0x929584 - std::sys_common::backtrace::_print_fmt::h621902e5e498623c
Oct 04 22:14:01 raspberrypi librespot[1063]:                        at src/libstd/sys_common/backtrace.rs:78
Oct 04 22:14:01 raspberrypi librespot[1063]:    3:   0x929584 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h39c43d9eea8febb9
Oct 04 22:14:01 raspberrypi librespot[1063]:                        at src/libstd/sys_common/backtrace.rs:59
Oct 04 22:14:01 raspberrypi librespot[1063]:    4:   0x950e88 - core::fmt::write::h1faa4c62c7847860
Oct 04 22:14:01 raspberrypi librespot[1063]:                        at src/libcore/fmt/mod.rs:1076
Oct 04 22:14:01 raspberrypi librespot[1063]:    5:   0x922cd4 - std::io::Write::write_fmt::h1d02362bb0653180
Oct 04 22:14:01 raspberrypi librespot[1063]:                        at src/libstd/io/mod.rs:1537
Oct 04 22:14:01 raspberrypi librespot[1063]:    6:   0x92be68 - std::sys_common::backtrace::_print::hb6e6bf916a89634f
Oct 04 22:14:01 raspberrypi librespot[1063]:                        at src/libstd/sys_common/backtrace.rs:62
[... cut ...]
Oct 04 22:14:03 raspberrypi librespot[1063]:                        at /build/glibc-FUvrFr/glibc-2.28/csu/libc-start.c:308
Oct 04 22:14:03 raspberrypi librespot[1063]: [2020-10-04T21:14:03Z ERROR librespot_playback::player] Player thread panicked!
Oct 04 22:14:03 raspberrypi systemd[1]: raspotify.service: Main process exited, code=exited, status=101/n/a
Oct 04 22:14:03 raspberrypi systemd[1]: raspotify.service: Failed with result 'exit-code'.
```

However, running /usr/bin/librespot directly on the terminal with the same arguments seems to work.

jake92918, 2020-12-27:
In case anyone has this error and doesn't want to waste 2 hours as I did. in the configuration file at /etc/default/raspotify under options I added this." --device hw:1,0 " where the 1 represents the card and the 0 represents the device
the information to find out what the number for your device and card is found by using the command " aplay -l "
so in the end your configuration file options should look like this.
OPTIONS="--username <USERNAME> --password <PASSWORD> --device hw:1,0"

closed 2021-11-24 by JasonLG1979