dtcooper/raspotify #206: Running Shairport and Raspotify

rhoamnasnen, 2019-05-17:
I was previously running shairport on my Model B2 and now added Raspotify to have a better experience on android (I still use shairport for e.g. Youtube on the Mac). 

**Problem**: The two Programms can't run simultaneously. The one, that is started first, works (and keeps working), while the other one shows that it's working (is connected and plays) without giving any output. Even after the music on the first one is pauses, or the first service is stopped completely. If you want to switch you have to kill the one you're using before starting the other one (that's a bummer). Yes, in both ways.

Has someone found a fix for that?

JonnieCache, 2019-05-18:
ive been running both and they work ok, even at the same time.

it could be that they are each opening the audio interface in some kind of exclusive mode, as opposed to using the alsa mixer. have a look in the configs to see how the audio interface is defined.

maximerenou, 2019-05-19:
Hello, are you running both as daemons ? It works for me. Good luck!

closed 2021-11-24 by github-actions[bot]