dtcooper/raspotify #207: Vero4k: since version 0.13.0 no sound and error in the logs

sanderegg, 2019-05-18:
I know it's maybe not officially supported on the Vero4k (running OSMC - debian based).

Since version 0.13.0 it fails with the following message after
```bash
systemctl status raspotify.service
```
```bash
 raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2019-05-18 18:56:37 CEST; 3min 28s ago
  Process: 2654 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 2651 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 2657 (librespot)
   CGroup: /system.slice/raspotify.service
           └─2657 /usr/bin/librespot --name raspotify (osmc) --backend alsa --bitrate 160 --disable-audio-cache --enable-volume-normalisation --linear-volume --initial-volume=100

May 18 18:56:42 osmc librespot[2657]: INFO:librespot_core::session: Authenticated as "slyou666" !
May 18 18:56:42 osmc librespot[2657]: INFO:librespot_core::session: Country: "CH"
May 18 18:56:42 osmc librespot[2657]: INFO:librespot_playback::audio_backend::alsa: Using alsa sink
May 18 18:56:43 osmc librespot[2657]: INFO:librespot_playback::player: Loading track "54-46 (Was My Number)" with Spotify URI "spotify:track:3UWidDMD5KcDBZevFGXm9Q"
May 18 18:56:43 osmc librespot[2657]: WARN:librespot_playback::player: Reducing normalisation factor to prevent clipping. Please add negative pregain to avoid.
May 18 18:56:44 osmc librespot[2657]: INFO:librespot_playback::player: Track "54-46 (Was My Number)" loaded
May 18 18:56:44 osmc librespot[2657]: ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function 'snd_pcm_hw_params_set_buffer_size_near' failed with error 'EINVAL: Invalid argument'
May 18 18:56:44 osmc librespot[2657]: ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
May 18 18:56:44 osmc librespot[2657]: ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function 'snd_pcm_hw_params_set_buffer_size_near' failed with error 'EINVAL: Invalid argument'
May 18 18:56:44 osmc librespot[2657]: ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
```

dtcooper, 2019-05-19:
Will take a look today. Thanks for the report.

spocon, 2019-07-17:
@sanderegg no sure if its not too late but you could check her. Seems the be the same problem: https://github.com/librespot-org/librespot/issues/333 otherwhise you could try this.:

https://spocon.github.io/spocon/

the backend is from the same guys which wrote the backend of raspotify but its the java version instead of rust

sanderegg, 2019-07-20:
yes I tried using pulseaudio. but then the sound was very bad in kodi... so I reverted

closed 2021-11-24 by github-actions[bot]