dtcooper/raspotify #208: Not discoverable on Windows 10, but is on Android

corriganc, 2019-05-25:
I've setup RaSpotify on my RPi as detailed in the instructions, and it works great (thank you for creating this!)

However, when I go into the "Connect to a device" window on my Windows 10 PC, the RaSpotify device does not show up. However, when I launch Spotify on my Android 9.0 device, the RaSpotify device shows up, allows me to connect and stream perfectly. The strange thing is, once I do this, Spotify on Windows 10 then realises the device exists and displays that it is being played on it, even letting me control it from Windows 10.

tl;dr - RaSpotify does not show up on Windows 10 but does on Android.

spocon, 2019-07-17:
Hi corriganc,
Could it be that your windows 10 is in another wlan or network ? subnetwork. I had the same issues and it fixed it. Otherwhise you could give this project a try https://spocon.github.io/spocon/

garethwelton, 2019-07-27:
I had a similar issue when not connected on mobile, then would not show on windows 10. I suspect this is a spotify issue.

andy646362, 2019-09-20:
Exactly the same problem here. Also, maybe unrelated, lots of general connection issues.

closed 2021-11-24 by github-actions[bot]