dtcooper/raspotify #210: Sound is Completely Corrupted

b-rad42, 2019-06-10:
So the first time I installed raspotify it worked perfectly for awhile and then stopped showing up as a spotify connect device.
I installed and reinstalled it and it again worked for awhile and then would show up as a device but just wouldn't connect.
Somehow it started working again when I rebooted my pi, but when I connected, the sound was completely corrupted and sounded like a vague demonic impression of the original song.
I tested to make sure my aux, amp, and speakers were working correctly by plugging it into my computer, and I tested my pi's soundcard by playing sounds on chromium; everything works perfectly. So the only possible problem is raspotify.
I uninstalled and removed the deb repo completely and reintalled, but to no avail (sound is still corrupted). The only possible reason I can think of is that my raspberry pi only has two bars of internet in my room (even though every other device has all bars), though I wouldn't expect that to corrupt the sound. I thought maybe it's because the bands it uses were too full with traffic, like it was on my chromecast audio, but again it just doesn't seem like that would corrupt the sound.
Thanks so much and have an amazing day,
Brad

spocon, 2019-07-17:
Hi b-rad42 , 

I had a similar issue , my raspotify was stopping on my raspberry pi b+ after 5 songs and i need to restart to get the next 4 songs. But that wasn't raspotifys fault , more a librespot bug ( the actual program behind raspotify). Therefore i decided to create my own project and use librespot-java instead librespot-rust. Until now i couldn't find any problems: https://spocon.github.io/spocon/

abcmer, 2019-09-10:
I also am experiencing this. I got raspotify working perfectly the first time. After rebooting my pi, the sound is completely corrupted (quarter speed, choppy, warped) but recognizable as the song I am playing. I have attempted restarting and reinstalling the service with no luck.

blaueente, 2019-10-03:
Same here. Note, I am using pulseaudio, via the librespot alsa driver that is redirected into pulseaudio. It would be great if your librespot binary would support pulseaudio directly. I tried to compile that, but I failed, because I didn't understand how to insert the pulseaudio libs into the process.

william-green, 2019-11-18:
I"m experiencing the same issue on Ubuntu Mate 18LTS for Raspberry pi. Raspberry pi 3 model b+.

abcmer, 2020-01-05:
I was able to fix by editing the following 2 properties in `/etc/pulse/daemon.conf`

```
avoid-resampling = yes
default-sample-rate = 48000
```

After restarting raspotify, the sound was no longer corrupted.

apnof, 2020-12-16:
for me, it works now by setting librespot option as follows:
**in file /etc/default/raspotify
`OPTIONS="--device hw:0,0"`**
(this is for 3,5mm analog audio jack output)

The suggestions regarding pulse daemon did not work for me.

JasonLG1979, 2021-11-17:
Raspotify is primarily intended to be used in a headless enviroment (Raspberry Pi OS lite) For desktop versions of Raspberry Pi OS (and/or systems with PulseAudio installed) [spotifyd](https://spotifyd.github.io/spotifyd/installation/Raspberry-Pi.html) offers similar functionality, much better PulseAudio compatibility and is a better choice.

closed 2021-11-17 by JasonLG1979