dtcooper/raspotify #213: Display album art?

alleybeachelectric, 2019-07-19:
I'd like to use a small TFT Display to display the album art while the track is playing (I'd be more than happy to pay for this feature). 
unfortunately, I'm pretty new to this type of stuff, but I was hoping to get some guidance. Thanks!
So far it's working great without any issues!

spocon, 2019-07-19:
Hi alleybeachelectric ,
Thank you for your feature request.
Librespot is a library which is used in raspotify. The scope there is not to have a GUI but have a connector for other devices. That said , I heard that librespot-java wants to write a API to get the latest song but than someone needs to write a front-end GUI for this . But that's another can of worms ....

aaron-rana, 2020-02-07:
any update on this?

map555, 2020-03-05:
@aaron-rana , @alleybeachelectric, as @spocon  said, the raspotify  program doesn't have a graphical user interface. In fact, it just display nothing. For what you are looking for, there is two options. The first one consist in looking on GitHub if somebody already do what you want. The second one consist in developping your own application. 

If you choose the second one, what thing you can do a graphical user interface for showing the song actually playing on your account and run raspotify on a second process who will run in parallel of  the gui one (but raspotify need sudo right...  :/ ). To be able to display actually playing song's informations, you will need to do requests to the Spotify API. From what I know there is actually there is no wrapper to do that. So you will have to do HTTP GET requests to the Spotify API in your favorite programming language. A request to the API returns a JSON dictionnary. So you will have to parse the JSON dictionnary to extract the informations you are looking for. For the album cover image, there is a list in the dictionnary. This list contains the urls for three size of the album cover image (the three sizes that Spotify uses in their apps: 640x640, 300x300, 64x64). To be able to show the album cover's image, you will have to load the image in your interface from one those urls. If you can successfully display the image, displaying the song's title, the artist's name(s) and the album's title will be an easy job for you. 

Also you will need a Spotify's token to do request to the api. If you are familiar with the Python programming language, there is a Python library for generating tokens for the Spotify's Web Player API. From what I know, each token is valid for one hour. So your program will have to get a key before the expiration of the actual one.

To resume, if you decide to do your own program, you will have to use tokens to do HTTP request to Spotify's API, parse the returned JSON to extract the informations of the song actually playing on your account and finally display the informations on the screen. You will have to use some timers to for repeating the HTTP Get request and the token generation at constant time intervals.

I invite you to consult the Spotify API's and the spotify-token's documentation.

Spotify API:
https://developer.spotify.com/documentation/web-api/reference/player/get-the-users-currently-playing-track/

spotify-token Python utility script:
https://pypi.org/project/spotify-token/

I hope this will help you

spocon, 2020-03-05:
I actually made a frontend on java / html basis.... but not for this project but for librespot-java backend. Its the same underlying engine than raspotify but implemented in java. It does the basic functions and it uses the REST API from librespot-java to get the informations:

Frontend : https://github.com/spocon/librespot-view

BUT: when i was finished with the first Alpha version i wanted to be more ambitous and have a native frontend X11 instead of a html frontend. So im currently focusing to implementing the frontend in python / X11 and will pack it in the spocon project (https://github.com/spocon/spocon) which uses librespot-java underneath (https://github.com/librespot-org/librespot-java)

Feel free to fork the frontend and continue if you like...

spocon, 2020-03-22:
FYI : 

There is a new project with a Spotify GUI introduced in a new project called [Lovspotify](https://github.com/spocon/lovspotifyl)  

For this i merged spocon and librespot-view and removed the project

I keep keep the spocon project alive but it doesn't include neither a GUI nor a REST-API

closed 2021-11-24 by github-actions[bot]