dtcooper/raspotify #215: Can't Connect to raspotify

Arusina, 2019-07-28:
When I start any Spotify App (web player, Android App or Windows 10) I can see the Spotify Connect location. When I select the raspotify, the Client just try to connect but nothing will happen. This is a brand new setup, nothing else installed on the pi. 
In very few cases it can connect and works, but after a while I can't control the raspotfy (change song/stop player etc.), sometimes it's just end playing. I couldn't find any related topic, so I created a new ticket. 

my setup:
- raspberry pi 2b
- hifiberry dac+
- dietpi image / raspbian (tried both, same error)

What I have done so far:

- Dietpi image / raspbian image
- install via dietpi / install via "Easy Installation" from Github
- restart the service
- use the login option in the config/without it
- change bitrate in config file (low, mid and high)
- change name of the device

When I run "sudo systemctl status raspotify" after I tried to connect, I get the following error
```
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2019-07-28 09:00:42 BST; 4min 24s ago
  Process: 1160 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 1157 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 1164 (librespot)
    Tasks: 3 (limit: 4915)
   CGroup: /system.slice/raspotify.service
           └─1164 /usr/bin/librespot --name raspotify_wohnzimmer --backend alsa --bitrate 320 --disable-audio-cache --enable-volume-normalisation --linear-volume --initial-volume=100 --username **** --password ****

Jul 28 09:04:50 DietPi librespot[1164]: WARN:librespot_connect::spirc: Skipping track "spotify:episode:7jB6D8ukPN0YM5uwroQuvO" at position [13] of 15
Jul 28 09:04:50 DietPi librespot[1164]: WARN:librespot_connect::spirc: Skipping track "spotify:episode:1n86aZD4LL5QsmQ60MKijw" at position [14] of 15
Jul 28 09:04:50 DietPi librespot[1164]: WARN:librespot_connect::spirc: Skipping track "spotify:episode:3ja1NKx2A6ZkuVKFU1DMbL" at position [0] of 15
Jul 28 09:04:50 DietPi librespot[1164]: WARN:librespot_connect::spirc: Skipping track "spotify:episode:079fzLZTIB3Z60BAoamiDD" at position [1] of 15
Jul 28 09:04:50 DietPi librespot[1164]: WARN:librespot_connect::spirc: Skipping track "spotify:episode:1xf2rLrd8Z6Mab8Yb12Aww" at position [2] of 15
Jul 28 09:04:50 DietPi librespot[1164]: WARN:librespot_connect::spirc: Skipping track "spotify:episode:3EY5BA7zYSm3yMu0UNOI2h" at position [3] of 15
Jul 28 09:04:50 DietPi librespot[1164]: WARN:librespot_connect::spirc: Skipping track "spotify:episode:4qyM5sTpLjS3l3CMtM7Wus" at position [4] of 15
Jul 28 09:04:50 DietPi librespot[1164]: WARN:librespot_connect::spirc: Skipping track "spotify:episode:35my4SpgawRBwNzsKpY7t6" at position [5] of 15
Jul 28 09:04:50 DietPi librespot[1164]: WARN:librespot_connect::spirc: Skipping track "spotify:episode:0rE9QfvV3IkD8auV3apLiF" at position [6] of 15
Jul 28 09:04:50 DietPi librespot[1164]: WARN:librespot_connect::spirc: Skipping track "spotify:episode:3ruUXtGom7uuzV3cCKtkcZ" at position [7] of 15

```
Here is the config file

```
# /etc/default/raspotify -- Arguments/configuration for librespot

# Device name on Spotify Connect
DEVICE_NAME="raspotify_wohnzimmer"

# Bitrate, one of 96 (low quality), 160 (default quality), or 320 (high quality)
BITRATE="320"

# Additional command line arguments for librespot can be set below.
# See `librespot -h` for more info. Make sure whatever arguments you specify
# aren't already covered by other variables in this file. (See the daemon's
# config at `/lib/systemd/system/raspotify.service` for more technical details.)
#
# To make your device visible on Spotify Connect across the Internet add your
# username and password which can be set via "Set device password", on your
# account settings, use `--username` and `--password`.
#
# To choose a different output device (ie a USB audio dongle or HDMI audio out),
# use `--device` with something like `--device hw:0,1`. Your mileage may vary.
#
OPTIONS="--username ****  --password ****"

# Uncomment to use a cache for downloaded audio files. Cache is disabled by
# default. It's best to leave this as-is if you want to use it, since
# permissions are properly set on the directory `/var/cache/raspotify'.
#CACHE_ARGS="--cache /var/cache/raspotify"

# By default, the volume normalization is enabled, add alternative volume
# arguments here if you'd like, but these should be fine.
#VOLUME_ARGS="--enable-volume-normalisation --linear-volume --initial-volume=100"

# Backend could be set to pipe here, but it's for very advanced use cases of
# librespot, so you shouldn't need to change this under normal circumstances.
#BACKEND_ARGS="--backend alsa"
```

I don't really understand the meaining of the errorcode. Thanks in advance for any help :)

miodzio4, 2019-08-05:
First of all many thanks for the app

I have very similar issue
After raspotify restart its visible on Android or iPhone but when I click on this it just shows "Connecting" and nothing happens or sometimes it connects and nothing happens

pi@raspberrypi:~ $ sudo service raspotify status
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2019-08-05 21:15:24 CEST; 5min ago
  Process: 934 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 931 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 936 (librespot)
    Tasks: 3 (limit: 4915)
   CGroup: /system.slice/raspotify.service
           └─936 /usr/bin/librespot --name raspotify --backend alsa --bitrate 160 --disable-audio-cache --enable-volume-normalisation -

sie 05 21:15:24 raspberrypi systemd[1]: Starting Raspotify...
sie 05 21:15:24 raspberrypi systemd[1]: Started Raspotify.
sie 05 21:15:24 raspberrypi librespot[936]: INFO:librespot: librespot (raspotify v0.13.1) 5d2cb32 (2019-04-16). Built on 2019-04-29. Bu
sie 05 21:15:24 raspberrypi librespot[936]: INFO:librespot_core::session: Connecting to AP "gew1-accesspoint-a-svlm.ap.spotify.com:4070
sie 05 21:15:24 raspberrypi librespot[936]: INFO:librespot_core::session: Authenticated as "miodzio4" !
sie 05 21:15:24 raspberrypi librespot[936]: INFO:librespot_playback::audio_backend::alsa: Using alsa sink
sie 05 21:15:24 raspberrypi librespot[936]: INFO:librespot_core::session: Country: "PL"
sie 05 21:17:46 raspberrypi librespot[936]: INFO:librespot_playback::player: Loading track "Quiero Saber" with Spotify URI "spotify:tra
lines 1-18/18 (END)

Config:
# /etc/default/raspotify -- Arguments/configuration for librespot

# Device name on Spotify Connect
DEVICE_NAME="raspotify"

# Bitrate, one of 96 (low quality), 160 (default quality), or 320 (high quality)
BITRATE="160"

# Additional command line arguments for librespot can be set below.
# See `librespot -h` for more info. Make sure whatever arguments you specify
# aren't already covered by other variables in this file. (See the daemon's
# config at `/lib/systemd/system/raspotify.service` for more technical details.)
#
# To make your device visible on Spotify Connect across the Internet add your
# username and password which can be set via "Set device password", on your
# account settings, use `--username` and `--password`.
#
# To choose a different output device (ie a USB audio dongle or HDMI audio out),
# use `--device` with something like `--device hw:0,1`. Your mileage may vary.
#
#OPTIONS="--username <USERNAME> --password <PASSWORD>"
OPTIONS="--username xxxx --password xxxx"
# Uncomment to use a cache for downloaded audio files. Cache is disabled by
# default. It's best to leave this as-is if you want to use it, since
# permissions are properly set on the directory `/var/cache/raspotify'.
#CACHE_ARGS="--cache /var/cache/raspotify"

# By default, the volume normalization is enabled, add alternative volume
# arguments here if you'd like, but these should be fine.
#VOLUME_ARGS="--enable-volume-normalisation --linear-volume --initial-volume=100"

# Backend could be set to pipe here, but it's for very advanced use cases of
# librespot, so you shouldn't need to change this under normal circumstances.
#BACKEND_ARGS="--backend alsa"

sound card: bcm2835 alsa (alsa mixer)
pi@raspberrypi:~ $ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a020d3
Serial		: 00000000c906db42

cat /sys/firmware/devicetree/base/model
Raspberry Pi 3 Model B Plus Rev 1.3

OS release:
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

danielcweber, 2019-08-07:
@Arusina Had the exact same problem. Got it working after clearing the playlist on the device and rebooting the player. Maybe even a couple of times. Make sure you play a proper song, no podcast or audio book. I know I had a podcast in my playlist, and the mentioning of "episode" in the log also hints to an issue here.

closed 2021-11-24 by github-actions[bot]