dtcooper/raspotify #216: Vero 4K+: No Audio Output via HDMI

Jorditzki, 2019-08-11:
I just installed Raspotify on the Vero 4K+. All worked fine, I can select the Player in my iPhone App and it switches the music playback to the Vero but I don't get any audio output.

I looked up my Audio Hardware configuration with aplay -l and edited the configuration file accordingly. After that hasn't done anything I tried out some hw:0,1 etc, but nothing happened.

The Status always looks something like this, with the same errors in the End.

`````
Process:` 9366 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 9363 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 9369 (librespot)
   CGroup: /system.slice/raspotify.service
           -9369 /usr/bin/librespot --name raspotify (madosmc) --backend alsa --bitrate 96 --disable-audio-cache --enable-volume-normalisation --linear-volume --initial-volume=100

Apr 24 00:57:23 madosmc librespot[9369]: INFO:librespot_playback::audio_backend::alsa: Using alsa sink
Apr 24 00:57:23 madosmc librespot[9369]: INFO:librespot_core::session: Country: "CZ"
Apr 24 00:57:24 madosmc librespot[9369]: INFO:librespot_playback::player: Loading track "Dreaming Deeper" with Spotify URI "spotify:track:4KPRqt53xnT2Hs2rDGKiRu"
Apr 24 00:57:25 madosmc librespot[9369]: INFO:librespot_playback::player: Track "Dreaming Deeper" loaded
Apr 24 00:57:25 madosmc librespot[9369]: ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function 'snd_pcm_hw_params_set_buffer_size_near' failed with error 'EINVAL: Invalid argument'
Apr 24 00:57:25 madosmc librespot[9369]: ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
Apr 24 00:57:25 madosmc librespot[9369]: ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function 'snd_pcm_hw_params_set_buffer_size_near' failed with error 'EINVAL: Invalid argument'
Apr 24 00:57:25 madosmc librespot[9369]: ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
Apr 24 00:57:26 madosmc librespot[9369]: ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function 'snd_pcm_hw_params_set_buffer_size_near' failed with error 'EINVAL: Invalid argument'
Apr 24 00:57:26 madosmc librespot[9369]: ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
```

sanderegg, 2019-12-07:
maybe nothing to do with it but I had an issue on a vero 4k:
https://github.com/dtcooper/raspotify/issues/207

reverting below 0.13.0 fixed the issue

bluenazgul, 2020-03-01:
Installed 0.13.0 on my Vero 4k Plus and had the same issue - uninstalled 0.13.0 and installed 0.12.0 now it is working without any issues on my Vero 4k Plus (i only wish i had an interface inside OSMC to controll raspotify there too or autostop raspotify playback when an movie starts on OSMC

Jorditzki, 2020-03-12:
I just tried again with the latest version but it still wont work. Contrary to that, version 0.12.0 works fine out of the box! Hope future versions might work again.
Thanks @sanderegg and @bluenazgul

closed 2020-03-12 by Jorditzki