dtcooper/raspotify #219: Spotify Connect keeps dropping and music stops

x0rtrunks, 2019-09-07:
The last week or so I haven't been able to get Spotify Connect to work reliably. The Pi shows up as a spotify connect device. It connects and will start playing a song, but then after a few seconds the music stops and the Pi will disappear from the devices list. Then a few seconds later it will show up again and you can reconnect. But every time you start to play music it stops after a few seconds. This behaviour has only started in the past week or two. It was working fine about a month ago when I first set up Raspotify.

I'm using a RPi3B+, wired network connection, and a hifiberry shield. Everything was freshly installed and updated about a month ago. As far as I can tell the network is fine. The Pi is still reachable the whole time. 

Any ideas? did Spotify change something again? Or is it just me?

Tyfui90, 2019-09-09:
I also experience the Spotify disappearing and reappearing in the spotify connect menu, but i dont experience it disconnecting after a couple of seconds.

However when I edited /etc/default/raspotify to add a new OPTION that pointed to a bad script (ie one with a critical error) then I would experience spotify disconnecting after about 1 second.

I'd recommend sharing your /etc/default/raspotify, and checking what you might be calling that is causing this issue. off the top of my head, maybe the hardware number of the hifiberry got changed during an update?

x0rtrunks, 2019-09-09:
I think I figured it out. I was using a 4gb SD card and a minimal debian buster install. I knew it was going to be a tight install but I thought it would be ok. Turns out I was running low on disk space. I cloned to a larger 16gb card and expanded the filesystem and now it appears to be working without dropping.

x0rtrunks, 2020-03-07:
Turns out this is not solved as I had suspected. It turns out that it is infact raspotify that is filling up the drive. After switching to a larger SD card it was fine for a while but has managed to consume all of the free space once again.

/var/cache/raspotify/files/ is getting filled with a bunch of folders and files usually less than 100mb.

[image]

Any ideas?

x0rtrunks, 2020-03-08:
Looks like it's related to this: https://github.com/dtcooper/raspotify/issues/144

closed 2021-11-24 by github-actions[bot]