dtcooper/raspotify #222: Mono Output Only

mjkpolo, 2019-10-06:
Hello,
My problem is that raspotify is only playing in mono output from my usb dac: Schiit Modi 3 to my preamp to my stereo jbl speakers.

I have replaced defaults.ctl.card 0, defaults.pcm.card 0
with
defaults.ctl.card 1, defaults.pcm.card 1

but my DAC still is only in mono. my amixer looks like this:
Simple mixer control 'PCM',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]

I have gone to alsamixer and it offers no controls for even volume.

What should I do?

mjkpolo, 2019-10-06:
Update: I even used speaker test with 2 channels and it worked! So this seems to be specific to raspotify

closed 2019-10-18 by mjkpolo

ikurek, 2019-11-01:
I'm having the same issue, my volume level fell down recently, so it wanted to see whats happening. Altrough I used `alsamixer` to set PCM volume to 100, it's still quiet. After running `amixer` I get the same result, indicating mono sound:
```
  Simple mixer control 'PCM',0
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback -10239 - 400
  Mono: Playback 400 [100%] [4.00dB] [on]```