dtcooper/raspotify #223: OSMC Raspotify minute's silence after switching from TV to raspotify

fjp, 2019-10-11:
I am using a raspberry 3B+ with a HiFiBerry Digi+ Pro which is connected over S/PDIF to my amplifier. The HDMI of the Pi is connected to a TV. On the Pi I use osmc VERSION_ID="2019.07-1" with raspotify (I don't know the version). 

The problem is when I stop watching TV, using the TVHeadend server running on the Pi, and try to play music in spotify. Doing this takes about a minute before I actually hear the sound. The output during this minute's silence is:

> sudo systemctl status raspotify
> ● raspotify.service - Raspotify
>    Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
>    Active: active (running) since Fri 2019-10-11 23:38:23 CEST; 17min ago
>   Process: 613 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
>   Process: 610 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
>  Main PID: 616 (librespot)
>    CGroup: /system.slice/raspotify.service
>            └─616 /usr/bin/librespot --name raspotify (osmc) --backend alsa --bitrate 320 --disable-audio-cache --enable-volume-normalisation --linear-volume --initial-volume=100 --device hw:0
> 
> Okt 11 23:55:29 osmc librespot[616]: ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'EBUSY: Device or resource busy'
> Okt 11 23:55:29 osmc librespot[616]: ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
> Okt 11 23:55:34 osmc librespot[616]: ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'EBUSY: Device or resource busy'
> Okt 11 23:55:34 osmc librespot[616]: ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
> Okt 11 23:55:39 osmc librespot[616]: ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'EBUSY: Device or resource busy'
> Okt 11 23:55:39 osmc librespot[616]: ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
> Okt 11 23:55:43 osmc librespot[616]: ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'EBUSY: Device or resource busy'
> Okt 11 23:55:43 osmc librespot[616]: ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
> Okt 11 23:55:48 osmc librespot[616]: ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'EBUSY: Device or resource busy'
> Okt 11 23:55:48 osmc librespot[616]: ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed

I use the `dtoverlay=hifiberry-digi-overlay` as a soundcard overlay in the pi config. Th config contains also `dtparam=audio=off`.

Obviously the device or resource seems to be busy and I am wondering how to avoid this problem. Do you have any idea what the problem might be?

In the OSMC System settings I use ALSA: snd_rpi_hifiberry_digi, S/PDIF as audio output device.

In the raspotify config `sudo vim /etc/default/raspotify` I added the following

```
BITRATE="320"
OPTIONS="--device hw:0"
```

However, I don't know exactly what `OPTIONS="--device hw:0"` does? Do I have to use `OPTIONS="--device hw:1"` or any other number? Do I have to add any other options to the raspotify config or the raspberry pi config?

I am also not sure if I run the latest version of raspotify and how to update it. What are the commands to find the version of raspotify and how do I update it? Do I need to just run the curl command again?

Please let me know how I can avoid the minute's silence after switching from watching TV to Raspotify? Thank you.

Hermanio, 2019-10-14:
Kodi should have a setting called "Keep audio device alive". If I remember correctly then changing this setting results in the audio device being released sooner. The delay may still be there, up to 5 seconds, and the success rate depends on nothing making sound in Kodi (sound effects when moving around UI also count).

If you can't see the setting then make sure to use the Expert options view in Kodi to make all settings available.

fjp, 2019-10-19:
Thank you @Hermanio for the hint. This seems to solve the problem 👍 
For anyone interested here is the [link](https://kodi.wiki/view/Settings/System/Audio#Keep_audio_device_alive) to the kodi documentation of that setting.

closed 2019-10-19 by fjp