dtcooper/raspotify #225: Random crash, no automatic restart

drakam, 2019-10-17:
Hello,

First thanks to the people who made rasportify, it's very useful. 

I have encounter seemingly random crashes of raspotify (or libresport ?), you can see the log below.

It also seems that rasportify does not restart automatically after the event and is therefore no longer available on the network.

note that I do an automatic restart of the service (with cron) everyday at 08:30 in order to mitigate this issue.

Has someone an idea of what is the issue here and how to fix it?

thanks a lot,

```
> oct 16 16:26:33 raspberrypi librespot[961]: INFO:librespot_playback::player: Loading track "Hard Times" with Spotify URI "spotify:track:0w5Bdu51Ka25Pf3hojsKHh"
oct 16 16:26:33 raspberrypi librespot[961]: INFO:librespot_playback::player: Track "Hard Times" loaded
oct 16 16:29:36 raspberrypi librespot[961]: INFO:librespot_playback::player: Loading track "Two High" with Spotify URI "spotify:track:5huOzlgCK7PsMSG4TFBvKY"
oct 16 16:29:36 raspberrypi librespot[961]: INFO:librespot_playback::player: Track "Two High" loaded
oct 16 16:33:09 raspberrypi librespot[961]: INFO:librespot_playback::player: Loading track "Black Skinhead" with Spotify URI "spotify:track:722tgOgdIbNe3BEyLnejw4"
oct 16 16:33:09 raspberrypi librespot[961]: INFO:librespot_playback::player: Track "Black Skinhead" loaded
oct 16 16:36:16 raspberrypi librespot[961]: WARN:librespot_connect::spirc: Skipping track "spotify:meta:page:1" at position [9] of 30
oct 16 16:36:17 raspberrypi librespot[961]: thread 'main' panicked at 'Spotify servers returned an error. Restart librespot.', core/src/mercury/mod.rs:193:13
oct 16 16:36:17 raspberrypi librespot[961]: stack backtrace:
oct 16 16:36:17 raspberrypi librespot[961]:    0:   0x964fa7 - std::sys::unix::backtrace::tracing::imp::unwind_backtrace::h71a4d52fb38cd7de
oct 16 16:36:17 raspberrypi librespot[961]:                        at src/libstd/sys/unix/backtrace/tracing/gcc_s.rs:39
oct 16 16:36:17 raspberrypi librespot[961]:    1:   0x96066f - std::sys_common::backtrace::_print::hce8fc7b28ee67394
oct 16 16:36:17 raspberrypi librespot[961]:                        at src/libstd/sys_common/backtrace.rs:70
oct 16 16:36:17 raspberrypi librespot[961]:    2:   0x963193 - std::panicking::default_hook::{{closure}}::hdaad4cfa8d5b3133
oct 16 16:36:17 raspberrypi librespot[961]:                        at src/libstd/sys_common/backtrace.rs:58
oct 16 16:36:17 raspberrypi librespot[961]:                        at src/libstd/panicking.rs:200
oct 16 16:36:17 raspberrypi librespot[961]:    3:   0x962dcf - std::panicking::default_hook::hfb698835042c0732
oct 16 16:36:17 raspberrypi librespot[961]:                        at src/libstd/panicking.rs:215
oct 16 16:36:17 raspberrypi librespot[961]:    4:   0x9638bb - std::panicking::rust_panic_with_hook::h25201d3872ea325e
oct 16 16:36:17 raspberrypi librespot[961]:                        at src/libstd/panicking.rs:478
oct 16 16:36:17 raspberrypi librespot[961]:    5:   0x62e71b - std::panicking::begin_panic::h8bda59fdc50f7b7d
oct 16 16:36:17 raspberrypi librespot[961]:    6:   0x5cebc3 - librespot_core::session::Session::dispatch::h36c3e63b99b71471
oct 16 16:36:17 raspberrypi librespot[961]:    7:   0x5cf6fb - <librespot_core::session::DispatchTask<S> as futures::future::Future>::poll::h63418ae49000df22
oct 16 16:36:17 raspberrypi librespot[961]:    8:   0x5b09cb - <futures::future::join::Join<A, B> as futures::future::Future>::poll::h2cfe0c95fbd4325c
oct 16 16:36:17 raspberrypi librespot[961]:    9:   0x605443 - <futures::future::map::Map<A, F> as futures::future::Future>::poll::ha302a77ea40df556
oct 16 16:36:17 raspberrypi librespot[961]:   10:   0x61d20b - <futures::future::map_err::MapErr<A, F> as futures::future::Future>::poll::h0c98e29ed863d730
oct 16 16:36:17 raspberrypi librespot[961]:   11:   0x887b9b - futures::task_impl::std::set::hde5e86308be3dd24
oct 16 16:36:17 raspberrypi librespot[961]:   12:   0x88514b - <futures::task_impl::Spawn<T>>::poll_future_notify::h04a94cce9611d0f3
oct 16 16:36:17 raspberrypi librespot[961]:   13:   0x8975ab - tokio_current_thread::CurrentRunner::set_spawn::h07fb0eeb48b16e9a
oct 16 16:36:17 raspberrypi librespot[961]:   14:   0x896873 - <tokio_current_thread::scheduler::Scheduler<U>>::tick::hab940f358a527526
oct 16 16:36:17 raspberrypi librespot[961]:   15:   0x89887f - <tokio_current_thread::Entered<'a, P>>::turn::h42b31b2eeca752f8
oct 16 16:36:17 raspberrypi librespot[961]:   16:   0x885973 - <std::thread::local::LocalKey<T>>::with::h893eaae605d98f0b
oct 16 16:36:17 raspberrypi librespot[961]:   17:   0x885297 - <std::thread::local::LocalKey<T>>::with::h41b7ebc6b3ba2dd6
oct 16 16:36:17 raspberrypi librespot[961]:   18:   0x890317 - tokio_core::reactor::Core::poll::h4b1f2e4d00adf0ec
oct 16 16:36:17 raspberrypi librespot[961]:   19:   0x49f28f - tokio_core::reactor::Core::run::h12e3b01e0232aff0
oct 16 16:36:17 raspberrypi librespot[961]:   20:   0x49dfb3 - librespot::main::h55151ad450023eb3
oct 16 16:36:17 raspberrypi librespot[961]:   21:   0x47bca3 - std::rt::lang_start::{{closure}}::h5d8a9a9c38c99203
oct 16 16:36:17 raspberrypi librespot[961]:   22:   0x9632e3 - std::panicking::try::do_call::h25e833b9f76f9c1b
oct 16 16:36:17 raspberrypi librespot[961]:                        at src/libstd/rt.rs:49
oct 16 16:36:17 raspberrypi librespot[961]:                        at src/libstd/panicking.rs:297
oct 16 16:36:17 raspberrypi librespot[961]:   23:   0x96b1c3 - __rust_maybe_catch_panic
oct 16 16:36:17 raspberrypi librespot[961]:                        at src/libpanic_unwind/lib.rs:87
oct 16 16:36:17 raspberrypi librespot[961]:   24:   0x963d4b - std::rt::lang_start_internal::haee47fdaef2afaf3
oct 16 16:36:17 raspberrypi librespot[961]:                        at src/libstd/panicking.rs:276
oct 16 16:36:17 raspberrypi librespot[961]:                        at src/libstd/panic.rs:388
oct 16 16:36:17 raspberrypi librespot[961]:                        at src/libstd/rt.rs:48
oct 16 16:36:17 raspberrypi librespot[961]:   25:   0x49edcf - main
oct 16 16:36:17 raspberrypi librespot[961]:   26: 0x76dc7677 - __libc_start_main
oct 17 08:30:01 raspberrypi systemd[1]: Stopping Raspotify...
oct 17 08:30:01 raspberrypi systemd[1]: Stopped Raspotify.
oct 17 08:30:01 raspberrypi systemd[1]: Starting Raspotify...
oct 17 08:30:01 raspberrypi systemd[1]: Started Raspotify.
```

ayodio, 2019-11-30:
Hi it seems i have the same problem, have found a way to fix it ?

drakam, 2019-11-30:
Nope, I did not witness it recently, doesn't mean it's gone. 
I'm more bothered with the  following crash these days :

thread '<unnamed>' panicked at 'called `Result::unwrap()` on an `Err` value: ChannelError', src/libcore/result.rs:997:5

I'm sad that raspotify/librespot is not really reliable but maybe it would be saddier if it did not exist at all^^

PeterPablo, 2019-12-07:
I'm seeing the same thing as drakam. Interestingly things ran smooth for hours on my previous installation (Raspbian Buster upgraded from Stretch), whereas I'm now running on a newly setup Raspbian Buster. I'm not familiar with the code base at all. Searching for a call to unwrap() didn't yield any result from the GitHub web interface. How shall we go about debugging this?

PeterPablo, 2019-12-07:
This issue was reported at https://github.com/librespot-org/librespot/issues/172 quite some time ago. From skimming over the ticket, I can't see a resolution.

closed 2021-11-24 by github-actions[bot]