dtcooper/raspotify #226: Some conflict with openvpn

daneboi72, 2019-10-23:
Basically, when openvpn is active on my rpi, raspotify is unavailable on my mac for spotify-connect. When I stop openvpn, it becomes available. Seems pretty consistent. Otherwise works fine. Have only been using for a day though. Not a big problem for me but it might help some people who are having usage issues. On raspbian buster with DE, pi4b.

vguttmann, 2020-07-17:
Could it be that openvpn and librespot (which raspotify is based on) are trying to use the same port? You could try to set up a secondary fixed IP address for openvpn to use to avoid such interferences.

djbarnat, 2020-07-25:
Have you got your authentication username a password setup in your config file for raspotify. i had a similar issue running openvpnand adding spotify account detail to the raspotify config resolved it. Not saying thats definitely it but worth a go.

closed 2021-11-24 by github-actions[bot]