dtcooper/raspotify #227: listen to only one song

HarryCover78, 2019-10-30:
Good morning all :)

Thank you for all the work already done on your project.

I come to you because my raspotify instalation works (RPI 3B+ BUSTER connected to a LG bluetooth soundbar via bluealsa), but just for one song :(
If I change the usual or if the song is finished and passes to the following, there is no sound ... :(
I can show you the result of any command to solve this problem, and I hope it will be useful to others :)

Yours truly.

PS: sorry for my english

HarryCover78, 2019-11-02:
my investigation allowed me to discover this after the end of a first single:

```
Nov 02 13:16:54 raspi librespot[507]: ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'ENODEV: No such
Nov 02 13:16:54 raspi librespot[507]: ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
Nov 02 13:16:59 raspi librespot[507]: ALSA lib bluealsa-pcm.c:763:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: PCM not found
```

I am always open to help :)

HarryCover78, 2019-11-02:
I have reviewed my file /etc/asound.conf :

```
defaults.bluealsa {
    interface "hci0"
    device "2C:2B:F9:73:96:2D"
    profile "a2dp"
}
pcm.soundbar {
    type plug
    slave.pcm {
        type bluealsa
        device "2C:2B:F9:73:96:2D"
        profile "a2dp"
    }
    hint {
        show on
        description "Bluetooth Audio ALSA Backend"
    }
}
```

and now I have this :

```
Nov 02 14:57:29 raspi librespot[501]:   10:   0x5c7bbb - librespot_playback::player::PlayerInternal::run::h30cd0a243402dad5
Nov 02 14:57:29 raspi librespot[501]:   11:   0x53f1df - std::sys_common::backtrace::__rust_begin_short_backtrace::h44c1b8da74887ab6
Nov 02 14:57:29 raspi librespot[501]:   12:   0x524213 - std::panicking::try::do_call::hba2a0cc58b861e70
Nov 02 14:57:29 raspi librespot[501]:   13:   0xa0a1c3 - __rust_maybe_catch_panic
Nov 02 14:57:29 raspi librespot[501]:                        at src/libpanic_unwind/lib.rs:87
Nov 02 14:57:29 raspi librespot[501]:   14:   0x542ef3 - <F as alloc::boxed::FnBox<A>>::call_box::h854a224d722fa450
Nov 02 14:57:29 raspi librespot[501]:   15:   0xa095af - std::sys::unix::thread::Thread::new::thread_start::h04f7e0d7a89cb7cf
Nov 02 14:57:29 raspi librespot[501]:                        at /rustc/fc50f328b0353b285421b8ff5d4100966387a997/src/liballoc/boxed.rs:759
Nov 02 14:57:29 raspi librespot[501]:                        at src/libstd/sys_common/thread.rs:14
Nov 02 14:57:29 raspi librespot[501]:                        at src/libstd/sys/unix/thread.rs:81

```

dtcooper, 2019-11-03:
Seems like a Bluetooth connectivity problem. Do players like SoX, etc work properly?

HarryCover78, 2019-11-03:
Hello,
Thank you for your question.

Yes Sox works.
The first song started is Ok, next song have no sound, but the progress bar shows how much time passes. :(

```
wget http://www.kozco.com/tech/piano2.wav
aplay -D bluealsa ./piano2.wav
```

is OK too :(

in doubt I resimplify my /etc/asound.conf:

```
defaults.bluealsa.service "org.bluealsa"
defaults.bluealsa.interface "hci0"
defaults.bluealsa.device "2C:2B:F9:73:96:2D"
defaults.bluealsa.profile "a2dp"
defaults.bluealsa.delay 10000
```

```
sudo nano /etc/default/raspotify
DEVICE_NAME="raspi"
BITRATE="320"
OPTIONS="--username kril78 --password Tchouc78 --device bluealsa"
CACHE_ARGS="--cache /tmp/raspotify"
VOLUME_ARGS="--enable-volume-normalisation --linear-volume --initial-volume=100"
BACKEND_ARGS="--backend alsa"
```
thank you for your interest.

HarryCover78, 2019-11-04:
Okey,

when I restart the bluealsa service, the music comes back!

`sudo systemctl restart bluealsa.service`

I am going to the Troubleshooting section of :

https://github.com/Arkq/bluez-alsa

> It is not possible to run more than one instance of the BlueALSA server per D-Bus interface. If one tries to run second instance, it would fail with the "Could not acquire D-Bus name: org.bluealsa" error message. This message might also appear when D-Bus policy does not allow acquiring "org.bluealsa" name for a particular user - by default only is allowed to start BlueALSA server.

Does Raspotify recreate a new instance of each song?

a1ex90, 2019-11-05:
Hey I'm not sure, if it's the same issue or a different one with a similar outcome:
But since recently my raspotify stops after one song, when choosing an album or a song of an album. Also it stops, when pushing the next song button. When playing a playlist, it works fine as ever. I've been using raspotify for a long time now and never experienced this issue before. My guess is it has something to do with changes spotify made, since I didn't change anything on my raspi. Would still very appreciate a fix.

HarryCover78, 2019-11-05:
So. I left a clean installation. I do not choose the solution bluealsa but puseaudio.
The result is the same. One song no more.

While all is well in jack.

Configuration problem? Compatibility with raspiand buster? I do not know where to look.

I think to put this problem aside. It's starting to take me too long

FKainka, 2020-01-05:
Hi, 
I stumbled over the same issues. After some days I was able to get it running by updating 
bluealsa to version 2.0.

You have to compile the newer Version manually. These instructions helped me:

https://github.com/Arkq/bluez-alsa/wiki/Installation-from-source
https://github.com/Arkq/bluez-alsa 

Hope that helps someone.

SaulGoodman1337, 2020-12-03:
> Hi,
> I stumbled over the same issues. After some days I was able to get it running by updating
> bluealsa to version 2.0.
>
> You have to compile the newer Version manually. These instructions helped me: [...]

thanks ! upgrading bluealsa solve my issue !

SaulGoodman1337, 2020-12-03:
i have found a similar problem here (https://github.com/dtcooper/raspotify/issues/227)... 

then i built bluez-alsa myself and .... it works !

The original Raspian version seems to cause problems. should this be included in the readme?

stefano-lupo, 2021-03-07:
I've no idea how related this is, but I was having the same issue _using the 3.5mm jack_ (and not bluetooth).

In any case nuking pulseaudio seemed to do the trick for me - this pi is old and unused so I've no idea if I installed that at some point / it came with raspbian? (See https://github.com/dtcooper/raspotify/issues/330#issuecomment-792377268 for more details)

JasonLG1979, 2021-11-17:
Raspotify is primarily intended to be used in a headless enviroment (Raspberry Pi OS lite) For desktop versions of Raspberry Pi OS (and/or systems with PulseAudio installed) [spotifyd](https://spotifyd.github.io/spotifyd/installation/Raspberry-Pi.html) offers similar functionality, much better PulseAudio compatibility and is a better choice.

closed 2021-11-17 by JasonLG1979