dtcooper/raspotify #228: Volume lever is lower, audio switched to mono channel

ikurek, 2019-11-01:
I'm running raspotify on 2 raspberries (model 3B+) and I've recently noticed, that my volume level got unusually low, altrough I used `alsamixer` to set PCM volume to 100. After running `amixer` I get the same result, indicating mono sound:

```
  Simple mixer control 'PCM',0
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback -10239 - 400
  Mono: Playback 400 [100%] [4.00dB] [on]
```

This happened on both of my evices, running completely standalone from eachother. The only thing I did with both of them was a regular software update, no other hardware/software changes that could possibly influence the problem.

I've seen [this issue](https://github.com/dtcooper/raspotify/issues/222), which seems to describe a similar problem, but for unknown reason it's marked as closed.

AInteriorB, 2019-11-13:
Do you use HDMI for audio output? If yes, you can ignore amixer's output, it's for the 3,5mm jack.

ikurek, 2019-11-13:
@AInteriorB No, audio goes thought 3,5mm jack, HDMI is disconnected and disabled with raspi-config.

AInteriorB, 2019-11-13:
You should use hdmi audio output or a seperate soundcard due to the poor audio quality of the Pi's 3.5mm jack.

Maybe this helps: [How to get stereo sound ? (solved)](https://www.raspberrypi.org/forums/viewtopic.php?t=18755)

closed 2021-11-24 by github-actions[bot]