dtcooper/raspotify #229: Add option to build from latest librespot commit on dev branch

xMordax, 2019-11-06:
Would it be possible to add to the build process a option which would allow a user to clone the latest commit on the ```dev``` branch?

IiroP, 2019-11-06:
The `build.sh` script takes librespot source from `librespot` folder (if exists). So if you clone it there and run script, it'll use that.
Of course adding an option to choose branch would be possible.

xMordax, 2019-11-07:
which folder do you mean and what do you need to edit?

IiroP, 2019-11-07:
Nah, I guess the easiest way would be to edit the `build.sh`. Just add something like this between rows 33 and 34 (after `cd librespot`):
```
git checkout dev
git pull
```
So it would switch to dev branch and update it before continuing the build.

_I haven't tested this personally, but I guess it should work_

unre4l, 2019-11-15:
I tried building dev and master but can not get it to build successfully. 
It always fails on the alsa Dockerfile step on `CC=arm-linux-gnueabihf-gcc ./configure ..`:

```Dockerfile
 RUN curl -O https://www.mirrorservice.org/sites/ftp.alsa-project.org/pub/lib/alsa-lib-1.0.29.tar.bz2 \
    && tar xvjf alsa-lib-1.0.29.tar.bz2 && cd alsa-lib-1.0.29 \
    && CC=arm-linux-gnueabihf-gcc ./configure --host=arm-linux-gnueabihf --disable-python \
        --prefix=/toolchain/rpi-tools/arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi/arm-bcm2708hardfp-linux-gnueabi/sysroot \
    && make -j $(nproc --all) && make install \
    && cd .. && rm -rf alsa-lib-1.0.29.tar.bz2 alsa-lib-1.0.29
```
Docker output from `make`:
```
...
checking whether make supports nested variables... yes
checking for arm-linux-gnueabihf-gcc... arm-linux-gnueabihf-gcc
checking whether the C compiler works... no
configure: error: in `/toolchain/alsa-lib-1.0.29':
configure: error: C compiler cannot create executables
See `config.log' for more details
The command '/bin/sh -c curl -O https://www.mirrorservice.org/sites/ftp.alsa-project.org/pub/lib/alsa-lib-1.0.29.tar.bz2     && tar xvjf alsa-lib-1.0.29.tar.bz2 && cd alsa-lib-1.0.29     && CC=arm-linux-gnueabihf-gcc ./configure --host=arm-linux-gnueabihf --disable-python         --prefix=/toolchain/rpi-tools/arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi/arm-bcm2708hardfp-linux-gnueabi/sysroot     && make -j $(nproc --all) && make install     && cd .. && rm -rf alsa-lib-1.0.29.tar.bz2 alsa-lib-1.0.29' returned a non-zero code: 77
Makefile:9: recipe for target 'raspotify_*.deb' failed
make: *** [raspotify_*.deb] Error 77
```

Form config.log it seems like architectures dont match:
`/toolchain/rpi-tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/arm-linux-gnueabihf-gcc: cannot execute binary file: Exec format error`

<details>
<summary>

`config.log` for more details:
</summary>

```
This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by alsa-lib configure 1.0.29, which was
generated by GNU Autoconf 2.69.  Invocation command line was

  $ ./configure --host=arm-linux-gnueabihf --disable-python --prefix=/toolchain/rpi-tools/arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi/arm-bcm2708hardfp-linux-gnueabi/sysroot

## --------- ##
## Platform. ##
## --------- ##

hostname = a2dca2b1a0ff
uname -m = armv7l
uname -r = 4.19.66-v7+
uname -s = Linux
uname -v = #1253 SMP Thu Aug 15 11:49:46 BST 2019

/usr/bin/uname -p = unknown
/bin/uname -X     = unknown

/bin/arch              = unknown
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /toolchain/rpi-tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/
PATH: /usr/local/sbin
PATH: /usr/local/bin
PATH: /usr/sbin
PATH: /usr/bin
PATH: /sbin
[... 325 lines cut ...]
sbindir='${exec_prefix}/sbin'
sharedstatedir='${prefix}/com'
sysconfdir='${prefix}/etc'
target_alias=''

## ----------- ##
## confdefs.h. ##
## ----------- ##

/* confdefs.h */
#define PACKAGE_NAME "alsa-lib"
#define PACKAGE_TARNAME "alsa-lib"
#define PACKAGE_VERSION "1.0.29"
#define PACKAGE_STRING "alsa-lib 1.0.29"
#define PACKAGE_BUGREPORT ""
#define PACKAGE_URL ""
#define PACKAGE "alsa-lib"
#define VERSION "1.0.29"

configure: exit 77
```

</details>

Maybe somebody has more knowledge about it? would like to use the podcast feature which was merged into librespot dev.

unre4l, 2019-11-16:
@xMordax  current build with dev librespot, can play podcasts: [raspotify_0.13.1-1-ga7274ae-dirty~librespot.20191112T230327Z.05b281f_armhf.deb.zip](https://github.com/dtcooper/raspotify/files/3854089/raspotify_0.13.1-1-ga7274ae-dirty.librespot.20191112T230327Z.05b281f_armhf.deb.zip)

l3d00m, 2019-11-22:
I think you don't have to change anything if you do a clean rebuild, since librespot now selected "dev" as the default branch you'll automatically use dev.

dtcooper, 2019-12-05:
I'll look into this. Thanks for reporting.

closed 2021-11-24 by github-actions[bot]