dtcooper/raspotify #231: Click noise between songs

AInteriorB, 2019-11-13:
If I skip to the next song, everything is smooth. But if I'm listening till the end of a song I hear a "click" in the pause to the next one.

AInteriorB, 2019-12-07:
The popping occurs after each song because the av-receiver detaches after the hdmi audio stream ends. One can hear a clicking noise when the new audio stream is establishing (next song). 

The solution is to play silent audio in the background so that the audio stream never ends actually. You can do that by adding a small systemd script:
/etc/systemd/system/silentaudio.service
```
[Unit]
Description=Remove that clicking noice between songs that Andy hated
After=raspotify.service

[Service]
ExecStart=/usr/bin/aplay -D default -t raw -r 44100 -c 2 -f S16_LE /dev/zero

[Install]
WantedBy=multi-user.target
```
Enable and start the script by
`systemctl enable silentaudio`
`systemctl start silentadii`

The required CPU power is negligible.

a-type, 2020-07-03:
Thanks for this -

When I have tried to do what you described, raspotify itself can't take control of the audio when I start streaming. I can 'play' music to the device, but no sound is heard.

I checked `/var/log/daemon.log` and see continual logging of:

```
Jul  3 20:42:45 raspberrypi librespot[460]: ALSA lib pcm_dmix.c:1071:(snd_pcm_dmix_open) unable to create IPC semaphore
Jul  3 20:42:45 raspberrypi librespot[460]: [2020-07-03T19:42:45Z ERROR librespot_playback::audio_backend::alsa] Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'EACCES: Permission denied'
Jul  3 20:42:45 raspberrypi librespot[460]: [2020-07-03T19:42:45Z ERROR librespot_playback::player] Could not start audio: Alsa error: PCM open failed
```

Any ideas about that? This is on a Raspberry PI Zero W with an I2S amplifier connected via GPIO to a speaker. If I don't run the 'background audio' service you describe, I can stream to the device fine (except for the speaker pops).

closed 2021-11-24 by github-actions[bot]