dtcooper/raspotify #233: Can't disable pwr mgmt on pizero

jshep321, 2019-11-17:
Hi,
In the troubleshooting section, there was a method to disable the pwr mgmt:
```
snd_soc_core.conf in /etc/modprobe.d with this line in:
options snd_soc_core pmdown_time -1

``
dmesg showed this error after reboot
snd_soc_core: `' invalid for parameter `pmdown_time'

I believe the correct file needs an equal sign!
options snd_soc_core pmdown_time =-1

vil1driver, 2020-02-02:
Hi,

I agree

jshep321, 2020-02-03:
I think it is underscore.  lsmod yields this:
`snd_soc_core          188416  3 snd_soc_bcm2835_i2s,snd_soc_hifiberry_dac,snd_soc_pcm5102a`

This conversation shows both!
[https://www.raspberrypi.org/forums/viewtopic.php?t=199318](url)
I'm unable to find any indication that this parameter is taken into account in the logs.  Will test (using my ears) when I have time.

jshep321, 2020-02-03:
To clarify: this method works well on my "audioinjector" i2s sound card.  It does not work on the cheap "speakerphat" devices.  They pop like mad at the beginning/end of the sound files...

closed 2021-11-24 by github-actions[bot]