dtcooper/raspotify #235: No sound via USB (tried existing solutions, no luck)

Elementaly, 2019-11-30:
I appreciate this has already been raised in other issues (speciifcally #40) but I've not been able to get anywhere with existing solutions.

Pi is connected to an SMSL AD18 USB DAC/amplifier. Sound works fine when using the browser. 

Onboard audio has been disabled using:
`blacklist snd_bcm2835`
under `/etc/modprobe.d/alsa-blacklist.conf`

Output of `aplay -l` is:

`**** List of PLAYBACK Hardware Devices ****
card 1: AMP [SMSL AD18 AMP], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #207 `

Raspotify config includes `--device hw:1,0` under the service config file ( /lib/systemd/system/raspotify.service)

However, when I try to play via Raspotify, I get the following when using `sudo systemctl status raspotify`

`raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2019-11-30 13:17:18 GMT; 14min ago
  Process: 1155 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 1156 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 1157 (librespot)
    Tasks: 3 (limit: 2200)
   Memory: 7.0M
   CGroup: /system.slice/raspotify.service
           └─1157 /usr/bin/librespot --name raspotify (raspberrypi) --backend alsa --bitrate 160 --device hw:1,0 --disable-audio-cache --enable-volume

Nov 30 13:31:59 raspberrypi librespot[1157]: ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function 'snd_pcm_hw_params_set_f
Nov 30 13:31:59 raspberrypi librespot[1157]: ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
Nov 30 13:32:00 raspberrypi librespot[1157]: INFO:librespot_playback::player: Loading track "Dreams - 2004 Remaster" with Spotify URI "spotify URI "spotify:track:0
Nov 30 13:32:00 raspberrypi librespot[1157]: INFO:librespot_playback::player: Track "Dreams - 2004 Remaster" loaded
Nov 30 13:32:00 raspberrypi librespot[1157]: ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function 'snd_pcm_hw_params_set_f
Nov 30 13:32:00 raspberrypi librespot[1157]: ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
Nov 30 13:32:00 raspberrypi librespot[1157]: ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function 'snd_pcm_hw_params_set_f
Nov 30 13:32:00 raspberrypi librespot[1157]: ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
Nov 30 13:32:05 raspberrypi librespot[1157]: ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function 'snd_pcm_hw_params_set_f
Nov 30 13:32:05 raspberrypi librespot[1157]: ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
`
I guess "snd_pcm_hw_params_set_f" is being clipped (this text is copied directly from the terminal in the pi) but I've no idea what it's clipping, so can't make sense of the error. Any help would be greatly appreciated.

Blubbaa, 2019-12-17:
I am having the exact same problem, using the Respeaker Mic Array v2.0 and a Raspberry Pi 3

Tyfui90, 2020-02-13:
Did you both try adding it as

     OPTIONS="--device hw:1,0"

and not just 

      --device hw:1,0

And once you've written the file, then running

    sudo systemctl restart raspotify

to restart spotify and then testing to see if the USB DAC was working?

Elementaly, 2020-03-03:
Thanks for your help - unfortunately when I do that Raspotify disappears from the list of available devices in the Spotify app.

old-square-eyes, 2020-05-03:
I have the same issue. Blacklisted internal and referenced external...

Fresh install of Buster, updated. My output of 

● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2020-05-03 02:57:26 BST; 1min 2s ago
  Process: 476 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 478 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 480 (librespot)
    Tasks: 4 (limit: 2200)
   Memory: 6.9M
   CGroup: /system.slice/raspotify.service
           └─480 /usr/bin/librespot --name raspotify (raspberrypi) --backend alsa --bitrate 320 --disable-audio-cache --enable-volume-normalisation --linear-volume --initial-volume=100 --device hw:0,1

May 03 02:58:20 raspberrypi librespot[480]: [2020-05-03T01:58:20Z INFO  librespot_playback::audio_backend::alsa] Using alsa sink
May 03 02:58:20 raspberrypi librespot[480]: [2020-05-03T01:58:20Z INFO  librespot_core::session] Country: "NZ"
May 03 02:58:21 raspberrypi librespot[480]: [2020-05-03T01:58:21Z INFO  librespot_playback::player] Loading <HEART ATTACK (feat. lau.ra)> with Spotify URI <spotify:track:4esIFtCJ2RUFbJpDGluVG4>
May 03 02:58:26 raspberrypi librespot[480]: [2020-05-03T01:58:26Z INFO  librespot_playback::player] <HEART ATTACK (feat. lau.ra)> loaded
May 03 02:58:26 raspberrypi librespot[480]: ALSA lib pcm_hw.c:1822:(_snd_pcm_hw_open) Invalid value for card
May 03 02:58:26 raspberrypi librespot[480]: [2020-05-03T01:58:26Z ERROR librespot_playback::audio_backend::alsa] Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'ENOENT: No such file or directory'
May 03 02:58:26 raspberrypi librespot[480]: [2020-05-03T01:58:26Z ERROR librespot_playback::player] Could not start audio: Alsa error: PCM open failed
May 03 02:58:26 raspberrypi librespot[480]: ALSA lib pcm_hw.c:1822:(_snd_pcm_hw_open) Invalid value for card
May 03 02:58:26 raspberrypi librespot[480]: [2020-05-03T01:58:26Z ERROR librespot_playback::audio_backend::alsa] Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'ENOENT: No such file or directory'
May 03 02:58:26 raspberrypi librespot[480]: [2020-05-03T01:58:26Z ERROR librespot_playback::player] Could not start audio: Alsa error: PCM open failed
~

old-square-eyes, 2020-05-03:
Oh wow. I got it working.

Used `--deviceplughw:1,0` instead of `--device hw:1,0` in `/etc/default/raspotify`

Elementaly, 2020-05-06:
Amazing, thanks. I don't have access to my Raspberry Pi at the moment, but will certainly try that when I do.

josiglo, 2020-11-13:
> Oh wow. I got it working.
> 
> Used `--deviceplughw:1,0` instead of `--device hw:1,0` in `/etc/default/raspotify`

Got it to work as well thanks to your hint. The correct argument is `--device plughw:1,0` though, for anyone wondering.

EdwardJB, 2021-02-03:
> Oh wow. I got it working.
> 
> Used `--deviceplughw:1,0` instead of `--device hw:1,0` in `/etc/default/raspotify`

Just to clarify, for me the magic argument was `--device plughw:1,0` - i.e. _with a space_.

JasSuri, 2021-05-04:
> > Oh wow. I got it working.
> > Used `--deviceplughw:1,0` instead of `--device hw:1,0` in `/etc/default/raspotify`
> 
> Just to clarify, for me the magic argument was `--device plughw:1,0` - i.e. _with a space_.

This worked for me!

closed 2021-11-24 by JasonLG1979