dtcooper/raspotify #236: No Audio When using Ethernet

rets34, 2019-12-01:
Raspberry Pi model 1 B.
Normally I’ve used a usb Wi-Fi module for my pin to connect my pin to my Wi-Fi, Raspotify has worked just fine after setting up. But when I switch to Ethernet instead of the Wi-Fi module, I can’t get audio from the device using Raspotify. Any other sounds come through just fine however.

pkfoury, 2019-12-07:
Same here on model 3.

beetleHun, 2020-01-15:
Hi Mate!

Could you take a look at your  `/boot/config.txt`?

I was in the same situation. Last month I switched to ethernet. Later I realized I have no sound from raspotify (everything else worked fine). I could connect to it from different devices, the progress bar showed that the music is being played, but without sound. It turned out it was a human error.

I started to debug what's going on, restarts, reboots, reinstalls didn't work, and then it came into my mind that I modified my `/boot/config.txt` as well when I changed to ethernet. Back then I copied a previous version of it which was faulty, it included the `hdmi_ignore_edid_audio=1` video option. Referring to the [docs](https://www.raspberrypi.org/documentation/configuration/config-txt/video.md) it says
> Setting hdmi_ignore_edid_audio to 1 pretends that all audio formats are unsupported by the display. This means ALSA will default to the analogue audio (headphone) jack.

As the librespot starts the raspotify with `--backend alsa` option by default, alsa has fallen back to the analog (jack) output, which was not used eventually.

I removed the `hdmi_ignore_edid_audio=1` option from the `/boot/config.txt` and voilá, works like charm.

Maybe your system has a different issue, but it's worth to take a quick look into your config.

Regards

closed 2021-11-24 by github-actions[bot]