dtcooper/raspotify #237: Solved music randomly stops playing mid-track

oboote, 2019-12-06:
Hi guys,
Been running raspotify for 2 years on two different setups (Raspbian and OSMC) and really struggled with the "music randomly stops, could be after 20 seconds, 2 songs, could be after 1 hour".

When it happens there's usually nothing in the logs - even with verbose logging, the logs just go:

```Nov 20 10:27:42 officepi librespot[21836]: TRACE:librespot_audio::fetch: chunk 100 / 103 complete
Nov 20 10:27:42 officepi librespot[21836]: TRACE:librespot_audio::fetch: requesting chunk 101
Nov 20 10:27:43 officepi librespot[21836]: TRACE:librespot_audio::fetch: chunk 101 / 103 complete
Nov 20 10:27:43 officepi librespot[21836]: TRACE:librespot_audio::fetch: requesting chunk 102
Nov 20 10:27:43 officepi librespot[21836]: TRACE:librespot_audio::fetch: chunk 102 / 103 complete
Nov 20 10:27:43 officepi librespot[21836]: DEBUG:librespot_audio::fetch: File 96e6fe448652629359c13496af3121320d846eaa complete
[music randomly stops, client disconnects, nothing further in logs]
[$sudo service raspotify restart]
Nov 20 10:28:01 officepi systemd[1]: Stopping Raspotify...
Nov 20 10:28:01 officepi systemd[1]: Stopped Raspotify.
Nov 20 10:28:01 officepi systemd[1]: Starting Raspotify...
Nov 20 10:28:01 officepi systemd[1]: Started Raspotify.```
_(Sometimes it also happens whilst the track is still loading; as in the requesting chunks process isn't finished yet)_ 

It seems to be related to running librespot through raspotify (sorry guys :/ ). I'm afraid my "fix" is literally to manually launch run librespot in a screen session:

Running in a screen session I've had music playing continuously for 2 days, no disconnections.

I'm now using the following script to start/restart/stop librespot:

```#!/bin/bash

launch_librespot () {
  screen -dmS librespot `which librespot` \
  --name osmc \
  --bitrate 320 \
  --initial-volume 50 \
  --enable-volume-normalisation \
  --disable-audio-cache \
  --verbose \
  #--username [USERNAME] \
  #--password [PASSWORD] \
  --backend alsa
  #--device plughw:0 \
  #--onevent /var/local/www/commandw/spotevent.sh
  echo "librespot started"
}

if [[ "$1" == "start" ]]; then

  if [[ -z `screen -ls | grep librespot` ]]; then
    launch_librespot
  else
    echo "librespot is already running"
  fi
  exit 1

elif [[ "$1" == "restart" ]]; then

  if [[ -z `screen -ls | grep librespot` ]]; then
    echo "librespot not running"
  else
    echo "restarting librespot"
    kill -9 `screen -ls librespot | grep librespot | cut -d "." -f 1`
    screen -wipe | 1>/dev/null
  fi

  launch_librespot
  exit 1

elif [[ "$1" == "stop" ]]; then
  if [[ -z `screen -ls | grep librespot` ]]; then
    echo "librespot not running"
  else
    echo "stopping librespot"
    kill -9 `screen -ls librespot | grep librespot | cut -d "." -f 1`
    screen -wipe | 1>/dev/null
  fi
  exit 1

else
  echo "function: ./librespot [start|restart|stop]"
fi```

I'd prefer to use raspotify as a proper system service so I'm sharing this more as an indication to help find the issue - aka launching librespot some other way.

**NOTE: THIS DOES -NOT- SOLVE LIBRESPOT CRASHING IF THE CONNECTION DIES
That crash however DOES have a log entry and CAN be watched/restarted with a cronjob - haven't gotten round to that yet but will build that in later.**

oboote, 2020-02-03:
I've added two more elif blocks for additional functionality:

**NOTE: screen -X hardcopy /path/to/file prints the current content of the screen session (ie. the librespot verbose output) to a file. To be nice to the SD I'm sending it to a tmpfs ram folder at /run/user/1000/; change that to another destination if you don't have that tmpfs location.**

"status" prints a tail of the current output of the screen session

`elif [[ "$1" == "status" ]]; then

  if [[ -z `screen -ls | grep librespot` ]]; then
    echo "librespot not running"
  else
    screen -S librespot -X hardcopy /run/user/1000/librespot
    tail /run/user/1000/librespot
  fi
  exit 1`

"cron" checks the current output (as per status) and searches for error messages, triggering a restart of the screen daemon if one is found (currently only recognises ConnectionReset; I'm waiting to identify other signs its crashed/disconnected).

`elif [[ "$1" == "cron" ]]; then

  if [[ -z `screen -ls | grep librespot` ]]; then
    echo "librespot not running"
  else
    screen -S librespot -X hardcopy /run/user/1000/librespot
    if [[ `tail /run/user/1000/librespot | grep -c ConnectionReset` != 0 ]]; then
      echo "ConnectionReset detected"
      "$0" restart
    fi

  fi
  exit 1`

YogaSakti, 2020-02-29:
how do I use this script??

aminm1364, 2020-11-24:
Hey everyone,
The same problem happens to me after 4 weeks of setting my Raspberry Pi4 raspotify up. 
I have also found two same trades which are still open since 2 years ago.

https://github.com/dtcooper/raspotify/issues/84
https://github.com/librespot-org/librespot/issues/215

Cheers.

rainer-schreiber, 2020-12-05:
I also had the problem that after a while the music just stopped playing.

I was pulling my hair off for one whole day, following different steps that are explained in different posts, and then I found out that after I touched my mouse, the sound came back :) 

So no need for installing a newer version of librespot, but just Disable Screen Blanking under Raspberry Pi Configuration -> Display tab. I use a Raspberry Pi 4 B+ in combination with a Hifiberry DAC2 HD, which obviously also slips into hibernation when the screen goes blank...

See also the last line in this comment: https://github.com/dtcooper/raspotify/issues/112#issuecomment-608361253

closed 2021-11-24 by JasonLG1979