dtcooper/raspotify #246: No sound

FernonE, 2020-02-10:
I have a clean install of NOOB raspberry pi 3B+ and spotify sees the raspberry connect system but there is no sound. I've tested some normal audio which works fine and the raspberry does produce sound. I've no clue where to start debugging this. 

P.s. I've tried to look through the other issues. But haven't found a solution.

spocon, 2020-02-10:
i have the same setup and use librespot-java. I wrote a installer for this : 

installation instruction is basically the same as with 

https://github.com/spocon/spocon

I didn't have to change anything and my spotify client connected and played my song right away without any changes.

FernonE, 2020-02-10:
I've replaced my OS with debian lite and this for some reason directly solved my problem. I still have no clue why it didn't work in the first place though...

closed 2020-02-10 by FernonE