dtcooper/raspotify #247: Release a new version built with the newest librespot

Leon5x, 2020-02-16:
@dtcooper Could you please release a new version which is built against the newest librespot version 0.1.1? 
tldr: Building myself fixed all connection problems for me.

Some time back the raspotify installation started to stop music playback randomly after 5 minutes to 2 hours. Also after being idle for some time I would not be able to start music on the pi. I would need to restart the service. Looking into the service logs it looked like the librespot error 104 caused the problem. They fixed the locking behaviour after this error in the last release.

So by building the package myself three days ago, I fixed the problem and it is working fine for three days in a row. 

It would be nice if you could release a new version so that everybody can benefit from it, I have seen a lot of issues from people with the same problem, maybe they can be fixed this easy :)

dtcooper, 2020-02-16:
Ah, thanks for the ping! Are you able to test a new version? I won't be near a pi for a few days but can build one now. :)

Leon5x, 2020-02-16:
Yes I can test a new version right away :)
The one i built had the version number:
raspotify_0.13.1-2-gff0d195_librespot.20200130T014147Z.3672214

dtcooper, 2020-02-16:
Since 3672214 is the most recent commit against librespot master, that you've tested this version is enough to go on for me to release now. Building it :)

dtcooper, 2020-02-16:
Released version 0.14.0. Give a change to test it if possible, @Leon5x. Thanks for the report!

dtcooper, 2020-02-16:
[`.deb` package here](https://github.com/dtcooper/raspotify/releases/download/0.14.0/raspotify_0.14.0.librespot.20200130T014147Z.3672214_armhf.deb)!

closed 2020-02-16 by dtcooper

Leon5x, 2020-02-16:
I installed it and will report after testing a few days if there are any problems.
Thanks a lot.

dtcooper, 2020-02-19:
Working okay? Still haven't gotten a chance to test myself. 😐

Leon5x, 2020-02-19:
Yes for me it is working much better than before. Today I had the first time that music stopped, again with error 104. But I could start the music again, without the need to restart the service, which is much better. 
If I can reproduce the error consistent I will file a issue in the librespot repo.