dtcooper/raspotify #251: Start Raspotify without sudo rights

map555, 2020-03-04:
Hi!
I tested raspotify some months ago and it just works fine.
Now, I'm trying to use raspotify into a university project but the problem is that raspotify needs the super user rights. Is there a way to run it without sudo?

Note: Sorry if my english is bad. It's not my first language.

spocon, 2020-03-17:
Hej map,

[SpoCon](https://github.com/spocon/spocon) is a wrapper for librespot-java 's library by Gianluca Altomani. Spocon is similar to Raspotify but with two major 

- it has librespot-java in the background 
- it includes a REST-API where you can get pictures, text , song title  of the current song aso. [here is the link](https://github.com/librespot-org/librespot-java/tree/dev/api)

spocon, 2020-03-18:
ah forgot the best reason... it doesn't need root access to start the application ;-)

closed 2021-11-24 by github-actions[bot]