dtcooper/raspotify #253: Can't play this right now

frode, 2020-03-11:
Running the latest version of raspotify on a RPi3.
Yesterday it was working, today its not working anymore.

Getting this message:

"Can't play this right now" and when i mouse hover the message it says "Spotify is not active on this device"

[image]

Have tried to check for updates on raspbian and rebooting does not help..

koglevee, 2020-03-15:
Experiencing 'similar' problems.  Worked the morning of 3/12.  Ceased before ~12:pm EDT.  Fed by Win10.  Broke on-the-fly after a long and illustrious career, so...  anyone @Spotify have some insight before I start tearing things apart??

dheckl, 2020-03-26:
I have the same issue and tested it with librespot standalone, but here is also the error. Are there any changes at Spotify which brakes librespot?

frode, 2020-08-24:
I have not been using raspotify for months now and I wont be able to participate in any future testing

fphammerle, 2021-01-31:
https://community.volumio.org/t/solved-spotify-connect-cant-play-this-right-now/6663/2 reports:
> Today, my devices show up in the devices list as “Can’t play this right now”. [...]
> For some reason, Spotify Connect does not work when playing podcasts! When switching to an actual song, the devices were available again!

I was able to select the librespot device after switching from a podcast to songs

glensc, 2021-11-17:
duplicate: https://github.com/dtcooper/raspotify/issues/385

closed 2021-11-17 by JasonLG1979

rogerjbos, 2021-11-24:
Same experience here.  Anyone figure out why playing songs works fine but playing podcasts doesn't work?  Not critical, but it would be nice to listen to podcasts as well.

JasonLG1979, 2021-11-24:
> Same experience here. Anyone figure out why playing songs works fine but playing podcasts doesn't work? Not critical, but it would be nice to listen to podcasts as well.

It's an upstream bug. It will likely get fixed when `librespot` finishes reverse engineering the latest Spotify playback API and starts supporting other formats besides just vorbis streams. It's not a Raspotify bug. Raspotify is just a packaged build of `librespot`. Please don't necro-bump closed issues with me too's.