dtcooper/raspotify #259: No sound while using AIY voice kit v1

rileyball, 2020-04-09:
I'm using the AIY voice kit from google and whenever I play sound through my raspberry Pi it doesn't play. I have tried to add the --device hw:0,0 qualifier to execstart command. Here is what aplay -l gives if thats any help

**** List of PLAYBACK Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-codec-0 []
  Subdevices: 0/1
  Subdevice #0: subdevice #0

And this is what sudo systemctl status raspotify shows

● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2020-04-09 21:26:29 BST; 47s ago
  Process: 10706 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 10707 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 10708 (librespot)
    Tasks: 4 (limit: 2200)
   Memory: 3.6M
   CGroup: /system.slice/raspotify.service
           └─10708 /usr/bin/librespot --name raspotify (picroft) --backend alsa --bitrate 160 --disable-audio-cache --enable-volume-normalisation --linear-volume --initial-volume=100 --device hw:0,1

Apr 09 21:26:51 picroft librespot[10708]: [2020-04-09T20:26:51Z ERROR librespot_playback::audio_backend::alsa] Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'ENOENT: No such file or directory'
Apr 09 21:26:51 picroft librespot[10708]: [2020-04-09T20:26:51Z ERROR librespot_playback::player] Could not start audio: Alsa error: PCM open failed
Apr 09 21:26:56 picroft librespot[10708]: [2020-04-09T20:26:56Z ERROR librespot_playback::audio_backend::alsa] Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'ENOENT: No such file or directory'
Apr 09 21:26:56 picroft librespot[10708]: [2020-04-09T20:26:56Z ERROR librespot_playback::player] Could not start audio: Alsa error: PCM open failed
Apr 09 21:27:01 picroft librespot[10708]: [2020-04-09T20:27:01Z ERROR librespot_playback::audio_backend::alsa] Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'ENOENT: No such file or directory'
Apr 09 21:27:01 picroft librespot[10708]: [2020-04-09T20:27:01Z ERROR librespot_playback::player] Could not start audio: Alsa error: PCM open failed
Apr 09 21:27:06 picroft librespot[10708]: [2020-04-09T20:27:06Z ERROR librespot_playback::audio_backend::alsa] Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'ENOENT: No such file or directory'
Apr 09 21:27:06 picroft librespot[10708]: [2020-04-09T20:27:06Z ERROR librespot_playback::player] Could not start audio: Alsa error: PCM open failed
Apr 09 21:27:11 picroft librespot[10708]: [2020-04-09T20:27:11Z ERROR librespot_playback::audio_backend::alsa] Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'ENOENT: No such file or directory'
Apr 09 21:27:11 picroft librespot[10708]: [2020-04-09T20:27:11Z ERROR librespot_playback::player] Could not start audio: Alsa error: PCM open failed

KevinVitale, 2020-06-09:
I'm also having the same issue. Is this on a Raspberry Pi 4 by chance?

shawaj, 2021-01-04:
This is interesting, mentions something needing changing in pulseaudio https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md#install-voice-bonnethat-packages

bhagan0402, 2021-01-06:
> This is interesting, mentions something needing changing in pulseaudio https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md#install-voice-bonnethat-packages
> 
> ```
> You may also need to disable module-suspend-on-idle PulseAudio module for the Voice HAT:
> 
> sudo sed -i -e "s/^load-module module-suspend-on-idle/#load-module module-suspend-on-idle/" /etc/pulse/default.pa
> ```

Did not work for me. Using RPi 3b+ and i get no output from anything and raspotify disconnects when I pause.

aguennegou, 2021-05-16:
Hey !
I have the same issue with my AIY Voice Kit 1... 
Did anyone find a way to use the raspberry pi + hat as a spotify connect device ?
Thanks

aguennegou, 2021-05-16:
> I'm also having the same issue. Is this on a Raspberry Pi 4 by chance?
> 
> **EDIT**: Not sure if this helps, sound _finally_ came out of the speaker after killing _pulseaudio_:
> 
> `$ pulseaudio -k`

Thanks ! It worked for me as well !

JasonLG1979, 2021-11-17:
Raspotify is primarily intended to be used in a headless enviroment (Raspberry Pi OS lite) For desktop versions of Raspberry Pi OS (and/or systems with PulseAudio installed) [spotifyd](https://spotifyd.github.io/spotifyd/installation/Raspberry-Pi.html) offers similar functionality, much better PulseAudio compatibility and is a better choice.

closed 2021-11-17 by JasonLG1979