dtcooper/raspotify #261: Spotify connect disappears from device list after a while

Ja83La, 2020-04-15:
Using the Spotify connect client by the Spotify App in the same network works flawlessly. 
After startup i can access to the spotify connect device via the web-api as well. 
But after a while the  device disapears from the device list in the web api an i am no longer able to control it via the web-api.
I can reactivate acces to the device by playing a song using the Spotify-App in the same network. But after a while it disappears again.

Setup: Raspian buster lite on a Pi zero w with a Phat DAC.

Maybe related to https://github.com/librespot-org/librespot/pull/376

KriTaeCotus, 2020-08-01:
Just wanted to +1 this issue. Yesterday after doing a totally clean Raspberry Pi OS install on my Zero W, I installed raspotify and edited etc/default/raspotify to: 

1. Add my Spotify login
2. Customize the Connect name
3. Set the bitrate to 320

The device stayed visible in Connect for at least eight hours but then disappeared. I was able to make it show up again by doing `systemctl restart raspotify`. It seems that's only a temporary fix - I had this issue before I did the system restore as well. 

Since me and @Ja83La are reporting this as an issue with a Zero W, is it possible that it's down to the hardware in some way? I'd appreciate hearing from others about raspotify uptime on the Pi 2 and up.

a-type, 2020-08-01:
I'm also using a Zero W as a homemade Spotify alarm clock.

I found that Spotify can consistently play in the morning (obviously after no other activity overnight) after I added my username/password to the Raspotify config. This seems to empower the underlying library to reach out to Spotify first and re-establish a connection? Not exactly sure what the difference is, but it worked for me.

Obviously, that solution isn't ideal if you're planning to make the device more accessible to non-technical users, but if you're planning to do that it's probably best to pursue an official device relationship with Spotify anyway.

KriTaeCotus, 2020-08-01:
Thanks - sounds like the OAuth authentication might be enhancing reliability to some extent? Seems it's still not perfect. I'd still love to hear about uptime on non-Zero hardware; it's not listed on the raspotify homepage as a 'tested' device.

The OAuth thing is probably too technical for me; diving into the raspotify config file was a big step for me so that tells you where I'm at.

I know this is probably not a great solution but I was thinking, is there a way to just set a cron to run at some interval that sends the `systemctrl restart raspotify` command? Doing that seems to kick it back into gear. If raspotify could restart _itself_  or refresh its authentication (assuming that's the problem) every so often that'd probably be ideal, assuming this is a viable solution.

Ja83La, 2020-08-01:
In my case, the problem occurred with and without username/password in the
config file. I also had the impression that it works more stable with the
username/password set. However, the problem occurred in both scenarios.

reframalo, 2020-10-14:
I have the same issue on Raspberry Pi 3b. 2 days ago I installed with the current rasbperry pi os. It worked for one day without problems. But today it disappeared. After restart of the pi, it shows up for a few seconds but then disappears. I have set up the pi with username/password

closed 2021-11-24 by github-actions[bot]