dtcooper/raspotify #263: raspotify doesn't support architecture 'arm64'

JonathanGirardeau, 2020-05-01:
I've installed Ubuntu 64 bits 18.04.4 LTS on my raspberry pi and I faced some issues to install raspotify on it:
`> curl -sL https://dtcooper.github.io/raspotify/install.sh | sh
Raspotify installer only runs on a Raspberry Pi`
`> sudo apt-get update
...
N: Skipping acquire of configured file 'main/binary-arm64/Packages' as repository 'https://dtcooper.github.io/raspotify raspotify InRelease' doesn't support architecture 'arm64'`

I found this article on the net but it seems a bit dirty:
https://thewayeye.net/posts/installing-raspotify-arm64/

Do you plan to make raspotify compatible with Ubuntu 64 bits ? 
Or do you have tips to make it run ?

spocon, 2020-05-01:
lovspotify does support arm64 

https://github.com/spocon/lovspotify

tswindell, 2021-03-09:
Just run the below before installation:
`dpkg --add-architecture armhf`

jodo234, 2021-09-22:
I successfully installed raspotify on my new RPi4 and I receive the same error/warning during apt-get update. Does this mean, I wont receive any updates unless I issued that command tswindell posted? Or will there be a arm64 release?

JasonLG1979, 2021-11-24:
I would love to see native arm64 builds. PR's welcome!!!

timfi-amsterdam, 2022-01-26:
Is anyone working on this? I'm not familiar with deb repos but after looking around a little bit I think it's not that hard.

JasonLG1979, 2022-01-26:
> Is anyone working on this? I'm not familiar with deb repos but after looking around a little bit I think it's not that hard.

Then get if a shot. Like I said PR's welcome.

timfi-amsterdam, 2022-01-26:
Great, I'm giving it a try

JasonLG1979, 2022-01-26:
> Great, I'm giving it a try

In some distant future when Pi Zero V2's aren't unobtainium we will be able to drop support for Pi 1's and Pi Zero V1.x's and we'll be able to build 32 bit (ARMv7) and 64 bit ARM packages with just vanilla Debian and not have to pull in the ARMv6 toolchain.

timfi-amsterdam, 2022-01-28:
@JasonLG1979 i've created a pull request. I was not sure how to test the actual repository though. Is there some documentation on this?

JasonLG1979, 2022-01-29:
> I was not sure how to test the actual repository though. Is there some documentation on this?

No, not really I'm not really sure myself? I'd also like an x86_64 package while you're at it if you could so that we only need to bug @dtcooper once to do the repo set up. I think that if we have armhf, arm64, and amd64 that should cover just about everyone.

closed 2022-01-29 by JasonLG1979

JasonLG1979, 2022-01-29:
Well as far as I can tell everything went off without a hitch. amd64 would be nice but it's not a must.

timfi-amsterdam, 2022-01-29:
Great! I'll have a look at amd64 when I find the time.

JasonLG1979, 2022-01-29:
> Great! I'll have a look at amd64 when I find the time.

Sounds good, no rush. Another more pressing concern is that the soon-ish to be merged `new-api` `libresopt` branch does not compile (for ARMv6 anyway) with our current toolchain. I have not tried since your commit so I'm not sure if it works for arm64 either? My hope is that it's just a matter of tweaking something small for it to work for ARMv6 otherwise we would have to drop ARMv6 support which would be a bummer for Pi Zero users but would simplify our build process since everything could come directly from Debian as it does with the arm64 build.

JasonLG1979, 2022-01-30:
@timfi-amsterdam never mind I copypasta'd your PR and hacked in x86_64 support.

timfi-amsterdam, 2022-01-30:
> Sounds good, no rush. Another more pressing concern is that the soon-ish to be merged `new-api` `libresopt` branch does not compile (for ARMv6 anyway) with our current toolchain. I have not tried [...]

I looked at the librespot examples and I think that as long as they support armv6 it won't be hard to port that. If they decide to drop support, it might be time to also just go that route?

JasonLG1979, 2022-01-30:
> I looked at the librespot examples and I think that as long as they support armv6 it won't be hard to port that. If they decide to drop support, it might be time to also just go that route?

The build scripts in the contrib folder aren't well maintained. Being in contrib they're not really used regularly during development and are not tested when changes are made like the core code base. I personally basically ignore everything in the contrib folder. I would not count on them actually working all of the time.

JasonLG1979, 2022-01-30:
In my mind we are basically the example of how to build `librespot` in the wild.

PR #507 (2022-01-29): add arm64 support
This closes #263