dtcooper/raspotify #267: Run a python script as soon as music is playing

20100times, 2020-05-05:
Hi everyone,

First of all thank you for your work, this tool is awesome.
An [issue](https://github.com/dtcooper/raspotify/issues/204) has already been opened on this topic but I can figure the answer out.

My problem is probably pretty simple: I would like to run a python script (to turn on my hifi system) as soon as a music is playing on the raspberry pi.

If anyone has an idea about that I would be great.
Any help is welcome! Thank you for that.

kloppertje, 2020-05-09:
I use this, which works fine for me:       
 with open('/proc/asound/card0/pcm0p/sub0/status') as myfile:
                if 'RUNNING' in myfile.read():
                        musicDetected = 1

I've added an IR LED to an RPi Zero W, which is controlled by LIRC to turn on the amplifier without needing a relay.

20100times, 2020-05-12:
Thanks for your help @kloppertje. I managed to do it in a quite different way. If it can help :

**1. Create a script (for instance in /etc/default/).**

`#!/bin/bash
if [ "$PLAYER_EVENT" = "start" ] || [ "$PLAYER_EVENT" = "stop" ]; then
  sudo python /home/pi/on.py
fi
`
Don't forget to do `sudo chmod 775 /path/to/your/script.sh` to make it executable.

**2. Modify the raspotify file.**

`sudo nano /etc/default/raspotify`
Then add the line
`OPTIONS="--onevent /path/to/your/script.sh"`

**3. Modify permission if needed**

If needed, add raspotify to sudo users with `sudo visudo`
At the very end of the file add `raspotify ALL = NOPASSWD: /usr/bin/python`.

NB : you can see the logs with `tail -f  /var/log/daemon.log`

Hope that it helps you !

benjaminburkhardt, 2020-05-13:
> Thanks for your help @kloppertje. I managed to do it in a quite different way. If it can help :
> 
> **1. Create a script (for instance in /etc/default/).**
> 
> ```
> if [ "$PLAYER_EVENT" = "start" ] || [ "$PLAYER_EVENT" = "stop" ]; then
>   sudo python /home/pi/on.py
> fi
> ```
> 
> Don't forget to do `sudo chmod 775 /path/to/your/script.sh` to make it executable.
> 
> **2. Modify the raspotify file.**
> 
> `sudo nano /etc/default/raspotify`
> 
> Then add the line `OPTIONS="--onevent /path/to/your/script.sh"`
> 
> **3. Modify permission if needed**
> 
> If needed, add raspotify to sudo users with `sudo visudo`
> At the very end of the file add `raspotify ALL = NOPASSWD: /usr/bin/python`.
> 
> NB : you can see the logs with `tail -f /var/log/daemon.log`
> 
> Hope that it helped you !

Thanks!
I did the same and controlled my stereo using a 433 Mhz power outlet. (https://github.com/ninjablocks/433Utils)

Works great!

vguttmann, 2020-07-17:
@20100times I think you can close this issue now.

closed 2020-07-20 by 20100times

farhandsome, 2020-10-28:
@20100times Thanks for the script, super helpful! Just a quick question. From a logic standpoint - won't your "on.py" script run everytime it stops as well? I figure you would want to run some sort of "off" script when the player stops.