dtcooper/raspotify #276: Cannot configure bluetooth device with raspotify

Aspyryan, 2020-06-07:
I tried alot already but can't get the connection to the bt device to work. 
When I configure everything like normal, the spotify connect plays through the monitor. 
So that's not what I want.

-  I tried changing the default backend from raspotify to bluealsa, but this didn't work, the service wouldn't start. 
-  Also changing our alsa configurations didn't work.
-  The connection to the bt device works fine when we try to play sound with aplay or youtube

I can provide additional information when needed

abragx, 2020-06-09:
Hi,
I doubt you can directly send to BT. 
I got BT + raspotify working with backend pipe and sox.

It should be possible to output directly to pulseaudio that has BT as sink. But looks like raspotify does not support backend pulseaudio.

abragx, 2020-06-10:
I recompiled librespot with pulseaudio support for raspberry PI and now it works pulseaudio and BT as sink.
Not using raspotify though but plain librespot.

Aspyryan, 2020-06-11:
Thanks abragx, gonna try

JasonLG1979, 2021-11-17:
Raspotify is primarily intended to be used in a headless enviroment (Raspberry Pi OS lite) For desktop versions of Raspberry Pi OS [spotifyd](https://spotifyd.github.io/spotifyd/installation/Raspberry-Pi.html) offers similar functionality, much better PulseAudio compatibility and is a better choice.

closed 2021-11-17 by JasonLG1979