dtcooper/raspotify #277: Raspotify losing connection after a while

Saik19922, 2020-06-13:
Rpi 3 running raspbian and Raspotify (2 other used webservers for home-network aswell if that's an issue).

Connecting, playing songs ect. works flawlessly but after a while it just disconnects from whatever device it was coupled to (in this case a smartphone that's on locked screen) and does it's own thing from there on. This is an issue since my GF likes going to bed with music and setting up a sleeptimer. I wanted to provide her a more smart option than plugging a cable in each day and a more customizable one than a 300$ bt speaker.

However sleep timer in the App just isn't working at all. I've setup a small Node.js server that directly speaks to the API and should pause the playback, however same behaviour. At 15 minutes it works, using a PC it works, at 60 even spotify itself says we're not playing anything (alas raspotify doing it's own thing somehow).

I'd love to not configure it for a specific account since we have seperate Spotify Accounts and I'm not using and web-interfaces ect. (like MusicPi) - just plain Connect.

nikolaskonto, 2020-06-13:
Hi, i think its an Android Spotify App issue. I have the same issue. I can't control the raspotify at a rpi3 with my phone the since last week or something. No issues using the windows app

Saik19922, 2020-06-13:
Well I figured my initial problem with the App I wrote was a Typo in my reauth with Spotify (Since the access token is only valid for 30 minutes) which broke it and inproper logging.

Controlling rpi3/raspotify with phone works well, it just desync's after a while. The App makes a sleeptimer work completely. I have it hosted on my github, all you really need is node/npm and pm2. Setup is pretty easy as well.

azuzunaga, 2020-08-16:
Having a similar issue with an iPhone. Raspotify works well for a bit and then stops playing audio. Sometimes it says it is still playing, sometimes the song completely stops and the spotify app looks like no song was selected. Restarting the service via `sudo systemctl restart spotify` fixes the issue. But clearly not a fix.

Saik19922, 2020-08-17:
@azuzunaga Can you specify more?

What behaviour do you expect and what do you get?

Does the Raspotify completely stop playback and lose connectivity to the Phone?
Does the same happen when using it with a PC/Not putting the Spotify App in background and not locking the phone?

I've since noticed that Raspotify needs a stable connection to check for the phone every so often. When you put spotify in the background on your phone however, that connection may not be able to establish anymore due to the phone having battery saving algorithms in place.

My main issue was that I was - at a certain point - unable to stop playback from phone (say a sleep timer, my girlfriend actually goes asleep with music and I didn't want to wake up at 3am because it's still playing/the spotify sleep timer failed because there was no active playback - atleast none the app knew of).

In the end you can still very easily control raspotify if it keeps playing after your phone lost connection. All you need is an access-/refresh-token pair and some fiddling with the spotify API. As I said I made a simple stop/sleep-timer node.js. I think I still have that on github, so if you know javascript/node.js and are interested in using that to emergency-control raspotify - you can go pretty ham with it.

To be exact you could even completely remove the need to access the Spotify App on your phone and control raspotify completely via Node.js. Maybe even attach 2 or 3 screens in the House and have that Webserver bookmarked/open on them to control multiple audio-synched raspberries (or single them out). I've done this with VMs and it's a pretty cool way to setup a DIY Home Audio system without the lack of rooms or something.

azuzunaga, 2020-08-17:
Hey! So I see the same behavior when playing from a computer. I press play, song maybe plays. Skip song, connection is broken and can't play any songs. `raspotify` service has to be restarted for it to work again.

I move a lot so I don't want to overcomplicate my setup. Playing from our phones was working well enough until a few weeks ago(?). I assume it was something on Spotify's end since nothing changed on the pi. Latest raspotify version has the same issues.

Since our problems don't seem to be related maybe this issue should be closed? I can open a new one just for my use case.

Thanks for your reply btw.

azuzunaga, 2020-08-23:
@JRansomed, I tried to repro the issue and I wasn't able to. Noticed that Spotify on my phone had been updated since I had the issues. I'll keep your fix in mind in case I run into this again. Thanks!

skywinder, 2020-08-31:
thanks for help,  @azuzunaga!

>I haven't checked in a while, maybe @dtcooper updated his repo. In the meantime, we compiled a new updated librespot. We've documented how we worked that out in that issue.

Can you provide the link to this doc and manual on how to update librespot? 

ps. when you talk "we" - you talking on behalf of librespot?

closed 2021-11-24 by github-actions[bot]

sysadminpower2019, 2023-08-04:
this is still an issue.