dtcooper/raspotify #281: Hide raspotify device from other members of a shared network.

Darkadrion, 2020-07-13:
Is there a way to hide the raspotify device from other devices in a shared network, so that only I can use it with spotify connect?
I connected raspotify to my spotify account, so I can controll it from anywhere as long as it's connected to ethernet. But I didn't find a way to hide it in the network, as my phone and computer are .

Thanks in advance.

vguttmann, 2020-07-17:
I think this is a thing that needs to be done in the device configuration itself, as it is on a windows PC. Though I do not know if/how that interferes with raspotify.

artmg, 2020-07-25:
I am not sure exactly what you mean by _hide_. Spotify Connect normally finds devices using mdns (aka bonjour or zeroconf). Librespot (which powers raspotify) advertises itself using this protocol, which is broadcast across your sub-network, making the device 'visible' to anything that browses (listens for) mdns. If you hide it then your Spotify client (on PC, phone or tablet) will not know that there is the Spotify Connect device available, so it will not appear under 'Devices Available'.

LibreSpot does have [an option](https://github.com/librespot-org/librespot/wiki/Options) `disable-discovery` but I'm not sure if raspotify exposes this nor exactly what effect it would have on a) your other network devices or b) your ability to Spotify Connect

dtcooper, 2020-07-26:
New version of librespot supports `--disable-discovery` now, so you can add that in `/etc/default/raspotify`.

closed 2020-07-26 by dtcooper